erpalma/throttled #255: Dell Latitude 5420 (i7-1185g7) Power limited

sameer, 2021-05-21:
Hi there,

First off just wanted to say thanks for creating this, it's a really helpful tool.

I recently set up throttled on a Latitude 5420 and am running into power throttling. cTDP-up gets set successfully but the package power still sticks at 15W. There are also errors around setting the temperature target and mchbar package power limit.

Is there something I can try to troubleshoot here? Let me know if there's any other information I should provide.

```
[AC] Thermal: OK - Power: LIM - Current: OK - Cross-domain (e.g. GPU): OK  ||  VCore: 762 mV - Package: 14.9 W - Graphics: 0.0 W - DRAM: 0.0 W
```
My config:
```
[GENERAL]
# Enable or disable the script execution
Enabled: True
# SYSFS path for checking if the system is running on AC power
Sysfs_Power_Path: /sys/class/power_supply/AC*/online
# Auto reload config on changes
Autoreload: True

## Settings to apply while connected to Battery power
[BATTERY]
# Update the registers every this many seconds
Update_Rate_s: 30
# Max package power for time window #1
PL1_Tdp_W: 29
# Time window #1 duration
PL1_Duration_s: 28
# Max package power for time window #2
PL2_Tdp_W: 44
# Time window #2 duration
PL2_Duration_S: 0.002
# Max allowed temperature before throttling
Trip_Temp_C: 85
# Set cTDP to normal=0, down=1 or up=2 (EXPERIMENTAL)
cTDP: 1
# Disable BDPROCHOT (EXPERIMENTAL)
Disable_BDPROCHOT: False

## Settings to apply while connected to AC power
[AC]
# Update the registers every this many seconds
Update_Rate_s: 5
# Max package power for time window #1
PL1_Tdp_W: 44
# Time window #1 duration
PL1_Duration_s: 28
# Max package power for time window #2
PL2_Tdp_W: 44
# Time window #2 duration
PL2_Duration_S: 0.002
# Max allowed temperature before throttling
Trip_Temp_C: 95
# Set HWP energy performance hints to 'performance' on high load (EXPERIMENTAL)
HWP_Mode: True
# Set cTDP to normal=0, down=1 or up=2 (EXPERIMENTAL)
cTDP:2
# Disable BDPROCHOT (EXPERIMENTAL)
Disable_BDPROCHOT: False
```

Debug logs:
```
[I] Detected CPU architecture: Intel TigerLake
[I] Trying to unlock MSR allow_writes.
[I] Loading config file.
[D] cpu platform info: maximum non turbo ratio = 18
[D] cpu platform info: maximum efficiency ratio = 4
[D] cpu platform info: minimum operating ratio = 4
[... 140 lines cut ...]
[D] core 7 thermal status: thermal limit status = 0
[D] core 7 thermal status: thermal limit log = 0
[D] core 7 thermal status: prochot or forcepr status = 0
[D] core 7 thermal status: prochot or forcepr log = 0
[D] core 7 thermal status: crit temp status = 0
[D] core 7 thermal status: crit temp log = 0
[D] core 7 thermal status: thermal threshold1 status = 0
[D] core 7 thermal status: thermal threshold1 log = 0
[D] core 7 thermal status: thermal threshold2 status = 0
[D] core 7 thermal status: thermal threshold2 log = 0
[D] core 7 thermal status: power limit status = 1
[D] core 7 thermal status: power limit log = 0
[D] core 7 thermal status: current limit status = 0
[D] core 7 thermal status: current limit log = 0
[D] core 7 thermal status: cross domain limit status = 0
[D] core 7 thermal status: cross domain limit log = 0
[D] core 7 thermal status: cpu temp = 33
[D] core 7 thermal status: temp resolution = 1
[D] core 7 thermal status: reading valid = 1
[D] TEMPERATURE_TARGET - write 0x5 - read 0x2 - match [91m[1mERR[0m
[D] CONFIG_TDP_CONTROL - write 0x2 - read 0x2 - match [92m[1mOK[0m
[D] MSR PACKAGE_POWER_LIMIT - write 0x42816000dd8160 - read 0x42816000dd8160 - match [92m[1mOK[0m
[D] MCHBAR PACKAGE_POWER_LIMIT - write 0x42816000dd8160 - read 0xffffffffffffffff - match [91m[1mERR[0m
[D] HWP - write "0x20" - read "0x20" - match [92m[1mOK[0m
```

erpalma, 2021-05-21:
I guess that your platform is limiting access to those MSRs.

sameer, 2021-05-21:
I read that undervolting was disabled to mitigate plundervolt on newer CPUs so I'm thinking that might be related. Or maybe there's some permanent BIOS setting that got carried over from Windows.

dcplaut, 2021-05-30:
I'm having similar issues (same processor in a Lenovo X1C9), including the match ERR for MCHBAR PACKAGE_POWER_LIMIT.  If I manually overwrite /sys/devices/virtual/powercap/intel-rapl-mmio/intel-rapl-mmio:0/constraint_0_power_limit_uw (as mentioned in the README under Static Fix), everything seems to work for a while, but eventually the embedded controller (or something) overwrites my change back to a lower value. Perhaps the mchbar_mmio.write32() calls (lines 740,741 in the script) need to be changed somehow.

phlb, 2021-06-05:
I've the same issue on XPS 13 9310 with i7-1185G7. 

```
[D] TEMPERATURE_TARGET - write 0x5 - read 0x0 - match ERR
[D] CONFIG_TDP_CONTROL - write 0x2 - read 0x2 - match OK
[D] MSR PACKAGE_POWER_LIMIT - write 0x42816000dd8160 - read 0x42816000dd8160 - match OK
[D] MCHBAR PACKAGE_POWER_LIMIT - write 0x42816000dd8160 - read 0xffffffffffffffff - match ERR
```

```
[AC] Thermal: OK - Power: LIM - Current: OK - Cross-domain (e.g. GPU): OK  ||  VCore: 735 mV - Package: 15.0 W - Graphics: 0.4 W - DRAM: 0.0 W  
```

Kernel:
```
Linux 5.10.0-7-amd64 #1 SMP Debian 5.10.40-1 (2021-05-28) x86_64 GNU/Linux
Kernel command line: BOOT_IMAGE=/vmlinuz-5.10.0-7-amd64 root=<...> ro lsm=capability,yama,apparmor,tomoyo msr.allow_writes=on i915.enable_dp_mst=0 quiet
```
Dell Bios Version :
```description: BIOS
vendor: Dell Inc.
physical id: 1
version: 2.2.0
date: 04/06/2021
```

Secure boot : 
```
sudo mokutil --sb-state
SecureBoot disabled

```

intel-rapl : 
```
sudo powercap-info -p intel-rapl
enabled: 1
Zone 0
  name: package-0
  enabled: 1
  max_energy_range_uj: 262143328850
  energy_uj: 22119838392
  Constraint 0
    name: long_term
    power_limit_uw: 44000000
    time_window_us: 27983872
    max_power_uw: 28000000
  Constraint 1
    name: short_term
    power_limit_uw: 44000000
    time_window_us: 2440
    max_power_uw: 0
  Constraint 2
    name: peak_power
    power_limit_uw: 105000000
    time_window_us: 0
    max_power_uw: 0
  Zone 0:0
    name: core
    enabled: 0
    max_energy_range_uj: 262143328850
    energy_uj: 17671414355
    Constraint 0
      name: long_term
      power_limit_uw: 0
      time_window_us: 976
  Zone 0:1
    name: uncore
    enabled: 0
    max_energy_range_uj: 262143328850
    energy_uj: 182139060
    Constraint 0
      name: long_term
      power_limit_uw: 0
      time_window_us: 976
Zone 1
  name: psys
  enabled: 0
  max_energy_range_uj: 262143328850
  energy_uj: 10893831425
  Constraint 0
    name: long_term
    power_limit_uw: 0
    time_window_us: 27983872
  Constraint 1
    name: short_term
    power_limit_uw: 97000000
    time_window_us: 976
```

academo, 2021-06-16:
I am having just this issue with the 1165G7  in a lenovo thinkpad.

similar to what @dcplaut mentioned, if I run the [static fix](https://github.com/erpalma/throttled#static-fix) it "releases" the power, but quickly something sets it back to 10w. 

What I ended up doing is a bash script that runs as a systemd service in an infinite loop every 5 seconds and set the values if is connected to AC only.

That fixes the problem for me but I can't stop thinking, what's exactly limiting the processor to 10W? Is it bad what I am doing? setting constantly a limit for 44w?

My computer runs very hot when I am doing heavy compilation things and I can see the temperature going in highs 90's and thermo throttling (which is good at that temperature!)

Perhaps someone has some light in this issue?

sameer, 2021-06-17:
> UPDATE: in a lenovo if you use Fn+l or Fn+m or Fn+h you can change the CPU performance mode.

Is that BIOS-level or is that something you're doing on Windows?

academo, 2021-06-23:
@sameer I do it in Ubuntu. Not sure if it can work in other distros.

dmirubtsov, 2021-07-03:
I think that it is Dell specific issue, not possible to fix it by 'static fix' on Latitude 7420/7520. Please correct me if I'm wrong

Related issue: https://github.com/intel/thermal_daemon/issues/293

sameer, 2021-07-04:
Thanks for sharing that, I think you're right.

dmirubtsov, 2021-07-04:
What I see currently:

1) In Windows it works correctly(3.3-4.8ghz)
2) Latest mainline kernel doesn't fix it
3) It depends on laptop vendor. I have MSI laptop with the same i7-1185g7 and it works properly, one difference is: Dell show `base_frequency` as 1800mhz, MSI as 3000mhz
```
```

dmirubtsov, 2021-07-08:
@sameer posted this issue on Dell Community forum: https://www.dell.com/community/Latitude/Latitude-5420-7420-7520-CPU-Throttling-Issue-on-Linux/m-p/7959019#M34444

please add details if you will find something, thanks

Yinette, 2021-07-23:
@academo For me (Lenovo P14s Gen 2) Turning off SpeedStep in the BIOS meant that it stopped giving the OS control of the CPU power (in 21.04's case I think it is thermald, I could be wrong...) It probably still needs to run the static fix on boot, I am not too sure yet. I'm working to determine this.

snoopcatt, 2021-07-27:
Hello, guys. Got same problem
My Latitude 7320 (i5-1135G7) goes 400MHz under any load. OS: Debian Buster.

With latest `throttled` it's possible to achieve only 1500 MHz.
Same result is with only `echo 0 > /sys/devices/virtual/powercap/intel_rapl.../enabled`.

Have you tried to boot Ubuntu 20.04 live?
it's kinda "certified for Dell" — https://ubuntu.com/certified/202010-28329

I haven't noticed any 400MHz drops while using LiveCD for an hour. 
But, installing kernel from Ubuntu to my main system didn't help at all.

dmirubtsov, 2021-07-27:
@snoopcatt 

Please check post in Dell community forum: https://www.dell.com/community/Latitude/Latitude-5420-7420-7520-CPU-Throttling-Issue-on-Linux/td-p/7959019

So far nothing new. Waiting for a miracle or fix from Dell

JMKS87, 2021-08-05:
Titular laptop & CPU Linux user also. I made some research & experiments.

In hot ambient (>25C) it behaves really bad, ie. stuck for long time at 1800MHz after some moderate load applied.
In colder ambient (~20C), even under full load, it maintained like 2400MHz.

But, I kind-of found the solution (or the culprit...?): when force-cooling the down side of the laptop (I'm using 120mm PC fan on an upside-down laptop, since I use exclusively external displays anyway), it stayed at about 3500MHz under load, even in 25C+ ambient.
And the cores were near the 100C themselves all the time. On the contrary, in throttling situations like above, they were at ~50C, but not speeding up despite that.
So, I think that there is some sensor **other than the CPU cores itself**, probably near the bottom, that induces that heavy CPU throttling.

My conclusion: should be fixable via BIOS, and I strongly hope that someone capable of it notices it too.

dmirubtsov, 2021-08-05:
@JMKS87 

Developer from `thermald` found that it related to TMEM sensor limit: https://github.com/intel/thermal_daemon/issues/293#issuecomment-873435404

So I think that you're right

dmirubtsov, 2021-08-05:
> My conclusion: should be fixable via BIOS

the weird thing is that it works correctly on Windows

JMKS87, 2021-08-07:
> > My conclusion: should be fixable via BIOS
> 
> the weird thing is that it works correctly on Windows

Ah, right. So, if Windows is not doing anything magical (I don't think it is, it's just more popular, so it's diagnosed & fixed much faster), Linux should be able to do it too, via software.
Or the case may be a little "speculative" - Windows has some workaround for ignoring some sensor, but since the software is closed-source, we don't know about it. And Linux "is not doing it" since it is not in line with guidelines.

In fact, tinkering with thermald's source code could be in reach for some of us, and maybe it is enough.

igoropaniuk, 2021-08-10:
@academo could you please share your `lenovo_fix.conf` you're currently using for your thinkpad ?

ftsogr, 2021-08-21:
I mentioned my temporary workaround some months ago in dell forums too...

I am using debian bullseye without **thermald** installed.
After the boot, I am running the commands below one by one and with the same sequence:

`sudo rmmod intel_rapl_msr
sudo rmmod processor_thermal_device
sudo rmmod processor_thermal_rapl
sudo rmmod intel_rapl_common
sudo rmmod intel_powerclamp

sudo modprobe intel_powerclamp
sudo modprobe intel_rapl_common
sudo modprobe processor_thermal_rapl
sudo modprobe processor_thermal_device
sudo modprobe intel_rapl_msr`

I am using these commands in a systemd service that I enabled it and it is running on boot.

n0rad, 2021-08-21:
thx @ftsogr 

I just tested on my latitude 7420 with **archlinux** and indeed without `thermald` nor `throttled` and just reloading the modules in order, it looks better.
I'm not stuck anymore at 400Mhz on load and going up to 4Ghz on normal usage.
but when stressing cores, it all get quickly stuck up to 1800Mhz

ftsogr, 2021-08-22:
I think that dell will take the issue more seriously if more people mention it in dell forums.

academo, 2021-09-01:
I'd like to update on this issue after using a lenovo laptop (P15s Gen2) with this processor for about 3 months.

Conditions:
* With AC plugged in (original charger, 65w)
* Using s-tui to monitor and stress
* Laptop set to performance with fn+h
* Thermold stopped, masked and uninstalled (you never know!)
* Latest BIOS available installed

When using high CPU this happens (no fixes applied):

* The laptop spikes up to 35w of power usage and after 1 or 2 seconds:
* It goes down to 10w and will not go beyond that. 1Ghz speed.

if the static fix mentioned in the readme is applied:

* The laptop spikes up to 35w of power usage and after 1 or 2 seconds:
* It goes down to 10w and will not go beyond that. 1Ghz speed.
* **Static Fix applied**
* It goes up to 35w again for about 3 to 5 seconds (seems random). Full clock speed.
* It goes down to 10w and will not go beyond that. 1Ghz speed.

When using throttled and performing a stress test, it behaves as it no fix was applied (no difference)

In practice:

* throttled itself doesn't fix the throttling problem for me
* The [static fix](https://github.com/erpalma/throttled#static-fix) does resolve the throttling for me but only for a short period of time (2-5 seconds).

My current solution?

Run the static fix in an infinite loop every 2 seconds. I added as a service for systemd. This is the script: https://gist.github.com/academo/30c781932b6b66adaacbc8932690f820

If any of you have a better solution or think what I am doing is dangerous for the CPU please let me know. I haven't found any other way to make this laptop work properly.

dmirubtsov, 2021-09-03:
I think that need to create separated issue for lenovo laptops because it's not related to throttling issue with dell

faustusdotbe, 2021-09-14:
> I am having just this issue with the 1165G7 in a lenovo thinkpad.
> 
> similar to what @dcplaut mentioned, if I run the [static fix](https://github.com/erpalma/throttled#static-fix) it "releases" the power, but quickly something sets it back to 10w.
> 
> What I ended up doing is a bash script that runs as a systemd service in an infinite loop every 5 seconds and set the values (the static fix commands) if is connected to AC only.
> 
> That fixes the problem for me but I can't stop thinking, what's exactly limiting the processor to 10W? Is it bad what I am doing? setting constantly a limit for 44w?
> 
> My computer runs very hot when I am doing heavy compilation things and I can see the temperature going in highs 90C degrees and thermo throttling (which is good at that temperature!)
> 
> Perhaps someone has some light in this issue?
> 
> UPDATE: in a lenovo if you use Fn+l or Fn+m or Fn+h you can change the CPU performance mode. This made my processor go from running in 10w to 20w but still not enough to make it to 44w so running the script still makes it go faster.

Hi!

Any chance you could share the script, as well as your config file? I have a P14s Gen2 and I'm being limited quite a lot. 

Additionally, what's the max frequency you get under a stress test? Mine does not go above 1.2 GHz under the `s-tui` stress test, despite temps being in the 50C range.

Thanks

academo, 2021-09-16:
@faustusdotbe the script is shared in my original message as a gist. Under a stress test I see all the cores going up 2.8Ghz and some boosting up beyond 3ghz.

Btw @whoenig created a new issue to follow up on the lenovos. let's move the conversation there. https://github.com/erpalma/throttled/issues/265

errogaht, 2021-11-26:
the same issue with i7-11850h but throttling going to 800MHz.

faustusdotbe, 2021-11-26:
Hi @errogaht 

If you have the thinkpad P14s please have a look here https://github.com/whoenig/thinkpad-p14s-gen2-ubuntu 

@whoenig compiled a few tricks to get the P14s working quite well. It's been my daily driver for a few months now and I've not encountered issues.

Grtschnk, 2021-11-29:
Same issue: Long time (+2s) Power throttling at 10-12W. It can boost but quickly goes down and stay around 1,800Mhz.
Latitude 7420, BIOS v1.22.2, Arch - up to date.

Also experiencing the issue with mismatching MSR writes and reads.

Not sure where exactly the fix is to be found: throttled, thermald, kernel, intel, dell or throttled..  I'll probably give up for now and just hope it will be fixed one day (only blaming the big vendors here, not any of the open source projects ;) ).

Grtschnk, 2021-11-29:
Wondering if this is related:
https://gist.github.com/Mnkai/5a8edd34bd949199224b33bd90b8c3d4
page 149 or search for PKG_PWR_LIM_LOCK https://www.intel.com/content/dam/www/public/us/en/documents/datasheets/10th-gen-core-families-datasheet-vol-2-datasheet.pdf

Now, is the register read correctly at all or is it only read only? If it is read correctly wouldn't that mean that the limit does not apply at all? 0xffffffff is a big number

kreuzman, 2021-12-17:
Thank @ftsogr I'm on Dell Latitude 5421 using using Debian Bullseye and loading/unloading of modules works for me. Like @n0rad when stressing cores to maximum frequency drops from Max Turbo to TDP frequency  (in my case 2.5 GHz). I suppose this is expected as Turbo Boost is not designed to run for long time on all cores.

In my case I needed to load/unload one more module **processor_thermal_device_pci_legacy** because it is blocking to rmmod **processor_thermal_device**. In my case order of models reloading looks like this (if it help to anyone).

`#!/bin/bash 

sudo rmmod intel_rapl_msr
sudo rmmod processor_thermal_device_pci_legacy
sudo rmmod processor_thermal_device
sudo rmmod processor_thermal_rapl
sudo rmmod intel_rapl_common
sudo rmmod intel_powerclamp

sudo modprobe intel_powerclamp
sudo modprobe intel_rapl_common
sudo modprobe processor_thermal_rapl
sudo modprobe processor_thermal_device
sudo modprobe processor_thermal_device_pci_legacy
sudo modprobe intel_rapl_msr`

sebastianha, 2021-12-17:
When you are doing this, do you stay stable for e.g. 10 minutes on 2.5GHz? At what temperatures?

kreuzman, 2021-12-17:
@sebastianha When running stress test it's usually in Turbo Boost for couple of minutes with ~3GHz (all cores) frequency with temp. ~90 °C. Then it's throttled to 2.5GHz (all cores) and it's stable with temp. ~75°C. Then after while it goes to Turbo Boost again. You can see running stress test on pictures.

https://imgur.com/a/HZZOmYa

sebastianha, 2021-12-17:
Sounds correct, thanks! Unfortunately this does not work on my 7320.

sebastianha, 2022-01-19:
> ```
> [D] TEMPERATURE_TARGET - write 0x5 - read 0x0 - match ERR
> [D] CONFIG_TDP_CONTROL - write 0x2 - read 0x2 - match OK
> [D] MSR PACKAGE_POWER_LIMIT - write 0x42816000dd8160 - read 0x42816000dd8160 - [...]

Might it be possible that writing to the MSRs is locked at all? It could be by accident that the second and third are ok as it had just the same value before. At least for me CONFIG_TDP_CONTROL is already set to 0x2 after boot.

sebastianha, 2022-01-20:
Dell is collection information: https://www.dell.com/community/Latitude/Latitude-5420-7420-7520-CPU-Throttling-Issue-on-Linux/m-p/8129749/highlight/true#M39458

sebastianha, 2022-03-10:
On thermald there just was a breakthrough for this problem: https://github.com/intel/thermal_daemon/issues/341#issuecomment-1063947568

Now the CPU runs at full power for 60s then throttles down (it does the same on Windows). Next steps are trying to remove this limitation, on Windows this is possible by locking MMIO via Throttlestop.

AdamantGarth, 2022-03-10:
It's weird, but it seems like the problem is fixed for me (on 5420) even without the patch. Compiling the kernel with it right now and the CPU is at 2.4GHz, 70°C for 25 minutes already.

spamik, 2022-03-10:
> On thermald there just was a breakthrough for this problem: [intel/thermal_daemon#341 (comment)](https://github.com/intel/thermal_daemon/issues/341#issuecomment-1063947568)
> 
> Now the CPU runs at full power for 60s then throttles down (it does the same on Windows). Next steps are trying to remove this limitation, on Windows this is possible by locking MMIO via Throttlestop.

I don't see this behaviour on my 7320. I've running this patch from morning, few minutes ago I've started system update (I have gentoo so compiling everything) and it's running fine - temperature near to 98°C all the time, frequency at start below 4GHz, after several minutes (5+) it downscales to something like 3,2GHz (temperature still on ~98°C) so I believe that it's totally OK for 3GHz base frequency CPU.

sebastianha, 2022-03-10:
There might be a difference as I am currently running on "optimized" BIOS settings. There is also "Ultraperformance" available but I currently have no time for intensive testing.

On 7320 the patch definitely fixed it. I have read in other tickets that this was not a problem on 5420 and others notebooks, that was also the reason I opened a ticket exactly for the 7320.

spamik, 2022-03-10:
Well, it may be this settings. I have ultra performance configured since I start solving this issue. I also want to try to set it back to optimized as now I can hear that this laptop is quite silent even running with fan on 100%. And with ultra performance it makes almost same noise under heavy load as with regular using so I want to keep it more silent on normal using. I'll let you know when I will have mood for reboot and reconfiguring BIOS it it's start throttling earlier.

lakotamm, 2022-08-03:
Would anyone with a Tigerlake CPU be able to run this command?
`sudo setpci -s 0:0.0 48.l`

It should report the address of the MCHBAR register. As @sameer  reported it seems like writes into the register fail. On Alder Lake we are experiencing the same issue and we found that the address has changed sometimes after the 10th gen and changing it to the correct one fixes the issue.

faustusdotbe, 2022-08-03:
```
❯ lscpu | grep "Model"
Model name:                      11th Gen Intel(R) Core(TM) i7-1165G7 @ 2.80GHz
Model:                           140
❯ sudo setpci -s 0:0.0 48.l
fedc0001
```

phlb, 2022-08-03:
```
> lscpu | grep "Model name"
Model name:                      11th Gen Intel(R) Core(TM) i7-1185G7 @ 3.00GHz
> setpci -s 0:0.0 48.l
fedc0001
```

lakotamm, 2022-08-03:
> ```
> ❯ lscpu | grep "Model name"
> Model name:                      11th Gen Intel(R) Core(TM) i7-1165G7 @ 2.80GHz
> ❯ sudo setpci -s 0:0.0 48.l
> fedc0001
> ```

Thanks! So the address is the same as for AlderLake. No wonder it does not work.

Could you please report cpuinfo?

`cat /proc/cpuinfo`

I enabled Tigerlake support in this commit and it would be great if anoyone could test it.

https://github.com/erpalma/throttled/pull/308/commits/532026d8994ffc70352ecc9b056e9f89c97ca0fd

faustusdotbe, 2022-08-03:
Sorry, the notification somehow went to spam. Find below the full `/proc/cpuinfo` . Many thanks for creating and linking the PR, I'll have a look in the coming days.

```
❯ cat /proc/cpuinfo

processor	: 0
vendor_id	: GenuineIntel
cpu family	: 6
model		: 140
model name	: 11th Gen Intel(R) Core(TM) i7-1165G7 @ 2.80GHz
stepping	: 1
microcode	: 0xa4
cpu MHz		: 2800.000
cache size	: 12288 KB
physical id	: 0
siblings	: 8
core id		: 0
cpu cores	: 4
apicid		: 0
initial apicid	: 0
fpu		: yes
fpu_exception	: yes
cpuid level	: 27
wp		: yes
flags		: fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush dts acpi mmx fxsr sse sse2 ss ht tm pbe syscall nx pdpe1gb rdtscp lm constant_tsc art arch_perfmon pebs bts rep_good nopl xtopology nonstop_tsc cpuid aperfmperf tsc_known_freq pni pclmulqdq dtes64 monitor ds_cpl vmx est tm2 ssse3 sdbg fma cx16 xtpr pdcm pcid sse4_1 sse4_2 x2apic movbe popcnt tsc_deadline_timer aes xsave avx f16c rdrand lahf_lm abm 3dnowprefetch cpuid_fault epb cat_l2 invpcid_single cdp_l2 ssbd ibrs ibpb stibp ibrs_enhanced tpr_shadow vnmi flexpriority ept vpid ept_ad fsgsbase tsc_adjust bmi1 avx2 smep bmi2 erms invpcid rdt_a avx512f avx512dq rdseed adx smap avx512ifma clflushopt clwb intel_pt avx512cd sha_ni avx512bw avx512vl xsaveopt xsavec xgetbv1 xsaves split_lock_detect dtherm ida arat pln pts hwp hwp_notify hwp_act_window hwp_epp hwp_pkg_req avx512vbmi umip pku ospke avx512_vbmi2 gfni vaes vpclmulqdq avx512_vnni avx512_bitalg avx512_vpopcntdq rdpid movdiri movdir64b fsrm avx512_vp2intersect md_clear flush_l1d arch_capabilities
vmx flags	: vnmi preemption_timer posted_intr invvpid ept_x_only ept_ad ept_1gb flexpriority apicv tsc_offset vtpr mtf vapic ept vpid unrestricted_guest vapic_reg vid ple pml ept_mode_based_exec tsc_scaling
bugs		: spectre_v1 spectre_v2 spec_store_bypass swapgs
bogomips	: 5606.40
clflush size	: 64
cache_alignment	: 64
address sizes	: 39 bits physical, 48 bits virtual
power management:

processor	: 1
vendor_id	: GenuineIntel
cpu family	: 6
model		: 140
model name	: 11th Gen Intel(R) Core(TM) i7-1165G7 @ 2.80GHz
stepping	: 1
microcode	: 0xa4
cpu MHz		: 2800.000
cache size	: 12288 KB
physical id	: 0
siblings	: 8
core id		: 1
cpu cores	: 4
apicid		: 2
initial apicid	: 2
fpu		: yes
fpu_exception	: yes
cpuid level	: 27
wp		: yes
flags		: fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush dts acpi mmx fxsr sse sse2 ss ht tm pbe syscall nx pdpe1gb rdtscp lm constant_tsc art arch_perfmon pebs bts rep_good nopl xtopology nonstop_tsc cpuid aperfmperf tsc_known_freq pni pclmulqdq dtes64 monitor ds_cpl vmx est tm2 ssse3 sdbg fma cx16 xtpr pdcm pcid sse4_1 sse4_2 x2apic movbe popcnt tsc_deadline_timer aes xsave avx f16c rdrand lahf_lm abm 3dnowprefetch cpuid_fault epb cat_l2 invpcid_single cdp_l2 ssbd ibrs ibpb stibp ibrs_enhanced tpr_shadow vnmi flexpriority ept vpid ept_ad fsgsbase tsc_adjust bmi1 avx2 smep bmi2 erms invpcid rdt_a avx512f avx512dq rdseed adx smap avx512ifma clflushopt clwb intel_pt avx512cd sha_ni avx512bw avx512vl xsaveopt xsavec xgetbv1 xsaves split_lock_detect dtherm ida arat pln pts hwp hwp_notify hwp_act_window hwp_epp hwp_pkg_req avx512vbmi umip pku ospke avx512_vbmi2 gfni vaes vpclmulqdq avx512_vnni avx512_bitalg avx512_vpopcntdq rdpid movdiri movdir64b fsrm avx512_vp2intersect md_clear flush_l1d arch_capabilities
vmx flags	: vnmi preemption_timer posted_intr invvpid ept_x_only ept_ad ept_1gb flexpriority apicv tsc_offset vtpr mtf vapic ept vpid unrestricted_guest vapic_reg vid ple pml ept_mode_based_exec tsc_scaling
bugs		: spectre_v1 spectre_v2 spec_store_bypass swapgs
bogomips	: 5606.40
clflush size	: 64
cache_alignment	: 64
address sizes	: 39 bits physical, 48 bits virtual
power management:

processor	: 2
vendor_id	: GenuineIntel
cpu family	: 6
model		: 140
model name	: 11th Gen Intel(R) Core(TM) i7-1165G7 @ 2.80GHz
stepping	: 1
microcode	: 0xa4
cpu MHz		: 2800.000
cache size	: 12288 KB
[... 134 lines cut ...]
model		: 140
model name	: 11th Gen Intel(R) Core(TM) i7-1165G7 @ 2.80GHz
stepping	: 1
microcode	: 0xa4
cpu MHz		: 2009.715
cache size	: 12288 KB
physical id	: 0
siblings	: 8
core id		: 3
cpu cores	: 4
apicid		: 7
initial apicid	: 7
fpu		: yes
fpu_exception	: yes
cpuid level	: 27
wp		: yes
flags		: fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush dts acpi mmx fxsr sse sse2 ss ht tm pbe syscall nx pdpe1gb rdtscp lm constant_tsc art arch_perfmon pebs bts rep_good nopl xtopology nonstop_tsc cpuid aperfmperf tsc_known_freq pni pclmulqdq dtes64 monitor ds_cpl vmx est tm2 ssse3 sdbg fma cx16 xtpr pdcm pcid sse4_1 sse4_2 x2apic movbe popcnt tsc_deadline_timer aes xsave avx f16c rdrand lahf_lm abm 3dnowprefetch cpuid_fault epb cat_l2 invpcid_single cdp_l2 ssbd ibrs ibpb stibp ibrs_enhanced tpr_shadow vnmi flexpriority ept vpid ept_ad fsgsbase tsc_adjust bmi1 avx2 smep bmi2 erms invpcid rdt_a avx512f avx512dq rdseed adx smap avx512ifma clflushopt clwb intel_pt avx512cd sha_ni avx512bw avx512vl xsaveopt xsavec xgetbv1 xsaves split_lock_detect dtherm ida arat pln pts hwp hwp_notify hwp_act_window hwp_epp hwp_pkg_req avx512vbmi umip pku ospke avx512_vbmi2 gfni vaes vpclmulqdq avx512_vnni avx512_bitalg avx512_vpopcntdq rdpid movdiri movdir64b fsrm avx512_vp2intersect md_clear flush_l1d arch_capabilities
vmx flags	: vnmi preemption_timer posted_intr invvpid ept_x_only ept_ad ept_1gb flexpriority apicv tsc_offset vtpr mtf vapic ept vpid unrestricted_guest vapic_reg vid ple pml ept_mode_based_exec tsc_scaling
bugs		: spectre_v1 spectre_v2 spec_store_bypass swapgs
bogomips	: 5606.40
clflush size	: 64
cache_alignment	: 64
address sizes	: 39 bits physical, 48 bits virtual
power management:
```

mateusz834, 2026-02-26:
I have have noted low `W` limits on this dell:

You can increase with these settigs:

```sh
echo 28000000 > /sys/class/powercap/intel-rapl-mmio/intel-rapl-mmio:0/constraint_0_power_limit_uw
echo 48000000 > /sys/class/powercap/intel-rapl-mmio/intel-rapl-mmio:0/constraint_1_power_limit_uw
echo 32000000 > /sys/class/powercap/intel-rapl-mmio/intel-rapl-mmio:0/constraint_1_time_window_us
```

This allows 28W and 48W peak during max 32s period.

Defaults are looow:

```
[mateusz@arch ~ ]$ cat /sys/class/powercap/intel-rapl-mmio/intel-rapl-mmio:0/constraint_0_power_limit_uw
15000000
[mateusz@arch ~ ]$ cat /sys/class/powercap/intel-rapl-mmio/intel-rapl-mmio:0/constraint_1_power_limit_uw
18750000
[mateusz@arch ~ ]$ cat /sys/class/powercap/intel-rapl-mmio/intel-rapl-mmio:0/constraint_1_time_window_us
2440
[mateusz@arch ~ ]$ cat /sys/class/powercap/intel-rapl-mmio/intel-rapl-mmio:0/constraint_0_time_window_us
27983872
[mateusz@arch ~ ]$
```