erpalma/throttled #258: Any drawback to setting update rate to 1 second?

0ki, 2021-07-06:
Is there any drawback to setting update rate to 1 second when on AC power?

With the default values of 5 seconds I frequently get throttled only now it fixed itself after a couple of seconds instead of the usual 5 minutes.

Great tool. Thanks!

erpalma, 2021-07-07:
Do you have any other userspace tool that are possibly interfering with throttled? Something like thermald or tlp?

Still, if you wish to lower the update rate to 1 second it should be fine. The only drawback is a very minor increase in CPU wakeups/sec on battery.

0ki, 2021-07-07:
I don't have tlp and I have disabled thermald per the instructions.

On battery I obviously care more about battery life than speed so I'm leaving 30 seconds there and 1 second on AC.
If what you say is accurate, shouldn't this ship with 1 second update rate on AC as the default?