erpalma/throttled #264: "'Update_Rate_s' parameter is missing"

zeldarocks2, 2021-09-05:
I have installed the throttled utility via 

`sudo dnf copr enable abn/throttled`
` sudo dnf install -y throttled`

on Fedora 34, I have modified the the throttled.conf file in /etc/ and started the systemd service using

`sudo systemctl enable --now throttled`

However, when I run 

`sudo throttled --monitor`

I get 

```
[I] Detected CPU architecture: Intel Kaby Lake (R)
[I] Loading config file.
[E] The mandatory "Update_Rate_s" parameter is missing.
```

When attempting to run the above with
`sudo throttled --config /etc/throttled.conf --monitor`

I get 

```
[I] Detected CPU architecture: Intel Kaby Lake (R)
[I] Loading config file.
Traceback (most recent call last):
  File "/usr/bin/throttled", line 882, in <module>
    main()
  File "/usr/bin/throttled", line 809, in main
    config = load_config()
  File "/usr/bin/throttled", line 374, in load_config
    config.read(args.config)
  File "/usr/lib64/python3.9/configparser.py", line 697, in read
    self._read(fp, filename)
  File "/usr/lib64/python3.9/configparser.py", line 1082, in _read
    raise MissingSectionHeaderError(fpname, lineno, line)
configparser.MissingSectionHeaderError: File contains no section headers.
file: '/etc/throttled.conf', line: 1
'\ufeff[GENERAL]\n'
```

erpalma, 2021-09-06:
I guess you broke the config file somehow while editing.

tienhuynh5312, 2021-11-30:
I ran the cmd under `--debug` flag and it said Secure boot should be disabled.

pi43r, 2022-07-22:
I'm having similar issues on Fedora36 with the copr release.

running `sudo throttled --monitor` gives me the above error message.

in `systemctl status throttled` however it succeeds:
```
● throttled.service - Stop Intel throttling
     Loaded: loaded (/usr/lib/systemd/system/throttled.service; enabled; vendor preset: disabled)
     Active: active (running) since Fri 2022-07-22 19:17:17 CEST; 17s ago
  [...]
Jul 22 19:17:17 t480s systemd[1]: Started throttled.service - Stop Intel throttling.
Jul 22 19:17:17 t480s throttled[26644]: [I] Detected CPU architecture: Intel Kabylake
Jul 22 19:17:17 t480s throttled[26644]: [I] Trying to unlock MSR allow_writes.
Jul 22 19:17:17 t480s throttled[26644]: [I] Loading config file.
Jul 22 19:17:17 t480s throttled[26644]: [I] Starting main loop.

```

and running `sudo throttled --config /etc/throttled.conf --monitor` also works fine