erpalma/throttled #265: Power Limited on ThinkPads (P14s Gen 2, P15s Gen2)

whoenig, 2021-09-08:
(Splitting up the related issue #255)

For Thinkpads with i7-1165G7 (so far confirmed P14s Gen 2, P15s Gen2), the `lenovo_fix.py --debug` script reports the following issue:

```
[D] MCHBAR PACKAGE_POWER_LIMIT - write 0x42816000dd8160 - read 0xffffffffffffffff - match ERR
```
In consequence, when using the `--monitor` flag it is visible that the CPU's power is frequently limited to 10 W. The static fix seems to work fine, but only for a limited time, see @academo description at https://github.com/erpalma/throttled/issues/255#issuecomment-910135681.

I have a P14s Gen 2 and can confirm that this is the same problem on that model as posted by @academo. I am using Ubuntu 20.04 with a 5.13.13 kernel.

FalsePositiv3, 2021-09-12:
I have exactly the same problem also on a Thinkpad T14 Gen2 with i7-1165G7 running arch linux

faustusdotbe, 2021-09-16:
Thanks for pointing me to here @academo. 

Using @academo's answer I managed to have a fix -- it's been running for a day now and works OK: basically a service running as root every 3 seconds that detects whether I am on AC power or not, and if yes that updates the values of the static fix. 

**NOTE**:
- Limit at 26W. Higher limits ([like the official 28W TDP](https://ark.intel.com/content/www/us/en/ark/products/208921/intel-core-i7-1165g7-processor-12m-cache-up-to-4-70-ghz-with-ipu.html)) brought it up to 100+C. 
- You need Intel Speedstep enabled in the bios

Below are the files if anyone's interested. 

- [POWERLIMIT.txt](https://github.com/erpalma/throttled/files/7175904/POWERLIMIT.txt)  --> rename as POWERLIMIT.sh, place in `/home/your-user/wattage`
- [wait-watt.txt](https://github.com/erpalma/throttled/files/7175905/wait-watt.txt) --> rename as wait-watt.sh, place in `/home/your-user/wattage`
[wattage .txt](https://github.com/erpalma/throttled/files/7175906/wattage.txt) --> rename as wattage.sh, place in `/etc/systemd/system`. Edit the user path on line 6 to reflect your own username (i.e. replace `simon` by your username).

Then it's a matter of enabling it as a service and launching it at boot time. To do so refer to here: https://www.shubhamdipt.com/blog/how-to-create-a-systemd-service-in-linux/

On a side note: I also updated my BIOS and now I have proper deep sleep in Ubuntu. The fans stop etc. [Have a look at this, this is a real game changer.](https://forums.lenovo.com/t5/ThinkPad-P-and-W-Series-Mobile-Workstations/P14s-Gen-2-enable-S3-suspend-support/m-p/5080140)

whoenig, 2021-09-16:
Works for me as well - thanks! I kept thermald disabled (as described in the readme on this project) and use https://github.com/vmatare/thinkfan in parallel to keep the fan noise reasonable.

faustusdotbe, 2021-09-16:
Ah perhaps I should have mentioned thinkfan as well -- that's what's being outputted on the GIF as well ("speed" and "level"). Could you share your config file? I unfortunately am on the wrong computer right now but will share mine as well. 

My GIF above displays the output of this script (https://gist.github.com/faustusdotbe/f6a983d5afbb69a33e7353bb1d71f339), which -- if you don't read French -- you need to put in `/usr/local/bin`. It's not professional or anything but it does the job, I usually launch it right when I start the computer.

erpalma, 2021-09-17:
@faustusdotbe It seems that you are hitting turbo boost only for a small time frame, isn't it?

faustusdotbe, 2021-09-17:
> @faustusdotbe It seems that you are hitting turbo boost only for a small time frame, isn't it?

Correct. The GIF wasn't taken under heavy load and I believe my limiting at 26W does not allow it to go much higher. 28W was already hitting 100C, which I didn't want to have. I could retry with a `s-tui` stress test if you want.

erpalma, 2021-09-17:
Oh I see. If you are hitting 100C then you are definitely limited by your cooling capacity rather than PL1/2.

mefistoso, 2021-09-22:
First, I want to thank you all for taking the time to debug this, I have been struggling with these issues already for a few days. It is nice to see that I'm not alone and that there are workarounds for this. Thinkpad P14s here running archlinux. 

So I tried what @academo suggested originally, just changing a bit of the watt values to 25000000 because I was already worried that my CPU would toast. I have a code running with multiple cores and the nvidia GPU at 100% so it is a pretty heavy stress. I do see temperatures of 95C or so, constantly. It is a bit worrisome but at least I can now run these things succesfully without the CPU throttling. The workaround/fix works, thanks again. I hope we can get a good clean way for solving this from Intel and/or Linux, eventually.

whoenig, 2021-09-24:
For anyone new who is struggling to set up the Thinkpad, I summarized my configuration of the Laptop (including fancontrol and the static fix described here) at https://github.com/whoenig/thinkpad-p14s-gen2-ubuntu. Comments/PRs etc. are welcome, of course.

academo, 2021-12-14:
there's a bios update from lenovo (1.46)  that allegedly fixes this issue.

I installed the update but could not get any performance improvement

the link to to the lenovo bios update page (see version 1.46 readme file) https://support.lenovo.com/us/en/downloads/ds548904-bios-update-utility-bootable-cd-for-windows-10-64-bit-thinkpad-p14s-gen-2-p15s-gen-2-t14-gen-2-t15-gen-2

faustusdotbe, 2021-12-14:
Thanks a lot for your edit, I was going to update the FW this afternoon..

academo, 2021-12-21:
@faustusdotbe I've been using the firmware for a while without problems with the fans.

fangfufu, 2022-01-16:
I seem to have similar issue:
https://github.com/erpalma/throttled/issues/291

lakotamm, 2022-08-03:
Would anyone with a Tigerlake CPU be able to run this command?
`sudo setpci -s 0:0.0 48.l`

It should report the address of the MCHBAR register. As @whoenig reported it seems like writes into the register fail. On Alder Lake we are experiencing the same issue and we found that the address has changed sometimes after the 10th gen and changing it to the correct one fixes the issue.

whoenig, 2022-08-03:
```
 ~ lscpu | grep "Model name"
Model name:                      11th Gen Intel(R) Core(TM) i7-1165G7 @ 2.80GHz
~ sudo setpci -s 0:0.0 48.l
fedc0001
~ cat /proc/cpuinfo
processor	: 0
vendor_id	: GenuineIntel
cpu family	: 6
model		: 140
model name	: 11th Gen Intel(R) Core(TM) i7-1165G7 @ 2.80GHz
stepping	: 1
microcode	: 0xa4
cpu MHz		: 2800.000
cache size	: 12288 KB
physical id	: 0
siblings	: 8
core id		: 0
cpu cores	: 4
apicid		: 0
initial apicid	: 0
fpu		: yes
fpu_exception	: yes
cpuid level	: 27
wp		: yes
flags		: fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush dts acpi mmx fxsr sse sse2 ss ht tm pbe syscall nx pdpe1gb rdtscp lm constant_tsc art arch_perfmon pebs bts rep_good nopl xtopology nonstop_tsc cpuid aperfmperf tsc_known_freq pni pclmulqdq dtes64 monitor ds_cpl vmx est tm2 ssse3 sdbg fma cx16 xtpr pdcm pcid sse4_1 sse4_2 x2apic movbe popcnt tsc_deadline_timer aes xsave avx f16c rdrand lahf_lm abm 3dnowprefetch cpuid_fault epb cat_l2 invpcid_single cdp_l2 ssbd ibrs ibpb stibp ibrs_enhanced tpr_shadow vnmi flexpriority ept vpid ept_ad fsgsbase tsc_adjust bmi1 avx2 smep bmi2 erms invpcid rdt_a avx512f avx512dq rdseed adx smap avx512ifma clflushopt clwb intel_pt avx512cd sha_ni avx512bw avx512vl xsaveopt xsavec xgetbv1 xsaves split_lock_detect dtherm ida arat pln pts hwp hwp_notify hwp_act_window hwp_epp hwp_pkg_req avx512vbmi umip pku ospke avx512_vbmi2 gfni vaes vpclmulqdq avx512_vnni avx512_bitalg avx512_vpopcntdq rdpid movdiri movdir64b fsrm avx512_vp2intersect md_clear flush_l1d arch_capabilities
vmx flags	: vnmi preemption_timer posted_intr invvpid ept_x_only ept_ad ept_1gb flexpriority apicv tsc_offset vtpr mtf vapic ept vpid unrestricted_guest vapic_reg vid ple pml ept_mode_based_exec tsc_scaling
bugs		: spectre_v1 spectre_v2 spec_store_bypass swapgs
bogomips	: 5606.40
clflush size	: 64
cache_alignment	: 64
address sizes	: 39 bits physical, 48 bits virtual
power management:

processor	: 1
vendor_id	: GenuineIntel
cpu family	: 6
model		: 140
model name	: 11th Gen Intel(R) Core(TM) i7-1165G7 @ 2.80GHz
stepping	: 1
microcode	: 0xa4
cpu MHz		: 2549.116
cache size	: 12288 KB
physical id	: 0
siblings	: 8
core id		: 1
cpu cores	: 4
apicid		: 2
initial apicid	: 2
fpu		: yes
fpu_exception	: yes
cpuid level	: 27
wp		: yes
flags		: fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush dts acpi mmx fxsr sse sse2 ss ht tm pbe syscall nx pdpe1gb rdtscp lm constant_tsc art arch_perfmon pebs bts rep_good nopl xtopology nonstop_tsc cpuid aperfmperf tsc_known_freq pni pclmulqdq dtes64 monitor ds_cpl vmx est tm2 ssse3 sdbg fma cx16 xtpr pdcm pcid sse4_1 sse4_2 x2apic movbe popcnt tsc_deadline_timer aes xsave avx f16c rdrand lahf_lm abm 3dnowprefetch cpuid_fault epb cat_l2 invpcid_single cdp_l2 ssbd ibrs ibpb stibp ibrs_enhanced tpr_shadow vnmi flexpriority ept vpid ept_ad fsgsbase tsc_adjust bmi1 avx2 smep bmi2 erms invpcid rdt_a avx512f avx512dq rdseed adx smap avx512ifma clflushopt clwb intel_pt avx512cd sha_ni avx512bw avx512vl xsaveopt xsavec xgetbv1 xsaves split_lock_detect dtherm ida arat pln pts hwp hwp_notify hwp_act_window hwp_epp hwp_pkg_req avx512vbmi umip pku ospke avx512_vbmi2 gfni vaes vpclmulqdq avx512_vnni avx512_bitalg avx512_vpopcntdq rdpid movdiri movdir64b fsrm avx512_vp2intersect md_clear flush_l1d arch_capabilities
vmx flags	: vnmi preemption_timer posted_intr invvpid ept_x_only ept_ad ept_1gb flexpriority apicv tsc_offset vtpr mtf vapic ept vpid unrestricted_guest vapic_reg vid ple pml ept_mode_based_exec tsc_scaling
bugs		: spectre_v1 spectre_v2 spec_store_bypass swapgs
bogomips	: 5606.40
clflush size	: 64
cache_alignment	: 64
address sizes	: 39 bits physical, 48 bits virtual
power management:

processor	: 2
vendor_id	: GenuineIntel
cpu family	: 6
model		: 140
model name	: 11th Gen Intel(R) Core(TM) i7-1165G7 @ 2.80GHz
stepping	: 1
microcode	: 0xa4
cpu MHz		: 2830.937
[... 135 lines cut ...]
model		: 140
model name	: 11th Gen Intel(R) Core(TM) i7-1165G7 @ 2.80GHz
stepping	: 1
microcode	: 0xa4
cpu MHz		: 2800.000
cache size	: 12288 KB
physical id	: 0
siblings	: 8
core id		: 3
cpu cores	: 4
apicid		: 7
initial apicid	: 7
fpu		: yes
fpu_exception	: yes
cpuid level	: 27
wp		: yes
flags		: fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush dts acpi mmx fxsr sse sse2 ss ht tm pbe syscall nx pdpe1gb rdtscp lm constant_tsc art arch_perfmon pebs bts rep_good nopl xtopology nonstop_tsc cpuid aperfmperf tsc_known_freq pni pclmulqdq dtes64 monitor ds_cpl vmx est tm2 ssse3 sdbg fma cx16 xtpr pdcm pcid sse4_1 sse4_2 x2apic movbe popcnt tsc_deadline_timer aes xsave avx f16c rdrand lahf_lm abm 3dnowprefetch cpuid_fault epb cat_l2 invpcid_single cdp_l2 ssbd ibrs ibpb stibp ibrs_enhanced tpr_shadow vnmi flexpriority ept vpid ept_ad fsgsbase tsc_adjust bmi1 avx2 smep bmi2 erms invpcid rdt_a avx512f avx512dq rdseed adx smap avx512ifma clflushopt clwb intel_pt avx512cd sha_ni avx512bw avx512vl xsaveopt xsavec xgetbv1 xsaves split_lock_detect dtherm ida arat pln pts hwp hwp_notify hwp_act_window hwp_epp hwp_pkg_req avx512vbmi umip pku ospke avx512_vbmi2 gfni vaes vpclmulqdq avx512_vnni avx512_bitalg avx512_vpopcntdq rdpid movdiri movdir64b fsrm avx512_vp2intersect md_clear flush_l1d arch_capabilities
vmx flags	: vnmi preemption_timer posted_intr invvpid ept_x_only ept_ad ept_1gb flexpriority apicv tsc_offset vtpr mtf vapic ept vpid unrestricted_guest vapic_reg vid ple pml ept_mode_based_exec tsc_scaling
bugs		: spectre_v1 spectre_v2 spec_store_bypass swapgs
bogomips	: 5606.40
clflush size	: 64
cache_alignment	: 64
address sizes	: 39 bits physical, 48 bits virtual
power management:
```

lakotamm, 2022-08-03:
Thanks! So the address is the same as for AlderLake. No wonder it does not work.

I enabled Tigerlake support in this commit and it would be great if you could test it.

[532026d](https://github.com/erpalma/throttled/commit/532026d8994ffc70352ecc9b056e9f89c97ca0fd)

Pull request #308

kylebakerio, 2022-11-14:
just want to mention that this seems to work on my tigerlake x1 extreme gen 4. Happy to do be walked through confirming it more directly, but before adjusting this I was seeing constant Power: LIM with the monitoring, and after adjusting config to basically set PL1 and PL2 to both be 100w and indefinite time, I now see that gone and power limit is only set according to thermals, which means I can sustain up to the low 50w's with fans on full blast, or high 40watts if on fan setting 7.

Likely able to do better if I disable the GPU, too, as the 3080 is using high 20 watts continuously.

fangfufu, 2023-04-25:
I think this issue can be closed. This is fixed via https://github.com/erpalma/throttled/pull/310.