erpalma/throttled #266: T480: Excuse my ignorance, but how do I tell if this is working (with s-tui)??

joaotavora, 2021-09-09:
I've a T480 that presumably suffers from this problem:
```
 sudo dmidecode | grep 'BIOS Revision'
        BIOS Revision: 1.28
```

I'm also on Arch.  I'm using

```
sudo systemctl start lenovo_fix.service
```

to enable and the corresponding `stop` to disable.  I don't see any difference in s-tui graphs.  I put `s-tui` into stress mode and see my CPU temps shoot to aroud 95 Celsius and the CPU freqs to around 3300Mhz then down to about 3050 Mhz.  Exactly the same is I stop or start the service.  What is going on: how does one see the throttling?

Arpafaucon, 2021-09-22:
I would think you do not suffer from this problem : on my computer the temperature does not go above 50°C, and the the frequency drops at 450-500 Mhz in s-tui in 'stress'  mode

joaotavora, 2021-09-22:
Thanks! Finally some clarity:) but then have I somehow escaped this bug?? This is how it always worked before throttled...

relativelycalmsea, 2022-12-03:
joaotavora,
I have a T480 as well, with the i5 8250U cpu. Throttled definitely works very well for me. Here is how you can see it in action:
Start/enable the service (sudo systemctl start throttled.service)
Or enable it to start automatically on boot (sudo systemctl enable throttled.service)
Or start it directly by navigating to where it's installed (probably /opt)
cd /opt/throttled and then sudo ./throttled.py
In another terminal start s-tui and put it to "stress." Watch your freqs and temps. For me, the freq. of all 4 cores goes to 3.4GHz and stays there for the duration of the test. Also, I see that the temp. of each core can exceed 80 degrees C, which before throttled, it was capped at 80 and could go no further. That is one way to see throttled work in real time.

May I suggest the tool powerstat? It is in most repos, and can give you a good indication of what your system's wattage is. I would start throttled, then start the stress test in s-tui, then in a new terminal start powerstat: sudo powerstat -R and let it scroll as the stress test is running. In my case, the package wattage hovered at just about 44W for the duration of the test.