erpalma/throttled #282: Why not adding support for Alder-Lake

djypku, 2021-12-12:
I guess there is no problem to use this program under 12th Core CPU, maybe it is not a good idea to block supporting ad hoc

erpalma, 2021-12-13:
You can easily test throttled on Alder Lake if you want.

mpawlowski-eyeo, 2022-07-25:
@erpalma I tested a Lenovo X1 Carbon 10th gen with an I7-1270P and `throttled` doesn't seem to work on it - the CPU pegs out at around 75 deg C, same as without `throttled`.

I'm running Kubuntu 22.04, disabled thermald and followed the instructions.

Anything I can try to get it running near its peak?

mpawlowski-eyeo, 2022-07-27:
I created a ticket for my specific CPU model: https://github.com/erpalma/throttled/issues/306

lakotamm, 2022-07-30:
i7-12700H is also reported as unsupported.

CicadaSeventeen, 2022-07-30:
Support or not is kind of ... complex? I also have 10750H device that not allow to use throttled. It depends on BIOS and OEM. So maybe we need more reported examples.

lakotamm, 2022-07-31:
I created a pull request + a fork where I bypassed the check so that we can run some tests.

From what I see decreasing TDP works, increasing does not.

lakotamm, 2022-07-31:
My observation:
PL-2 seems to work well. PL1 is limited to 20W both on AC an BAT. I have no clue why this is the case, especially since it does not occur in windows.

OS: Fedora 36
i7-12700H

[D] core 19 thermal status: thermal limit status = 0
[D] core 19 thermal status: thermal limit log = 0
[D] core 19 thermal status: prochot or forcepr status = 0
[D] core 19 thermal status: prochot or forcepr log = 0
[D] core 19 thermal status: crit temp status = 0
[D] core 19 thermal status: crit temp log = 0
[D] core 19 thermal status: thermal threshold1 status = 0
[D] core 19 thermal status: thermal threshold1 log = 0
[D] core 19 thermal status: thermal threshold2 status = 0
[D] core 19 thermal status: thermal threshold2 log = 0
[D] core 19 thermal status: power limit status = 0
[D] core 19 thermal status: power limit log = 0
[D] core 19 thermal status: current limit status = 0
[D] core 19 thermal status: current limit log = 0
[D] core 19 thermal status: cross domain limit status = 0
[D] core 19 thermal status: cross domain limit log = 0
[D] core 19 thermal status: cpu temp = 60
[D] core 19 thermal status: temp resolution = 1
[D] core 19 thermal status: reading valid = 1
[D] TEMPERATURE_TARGET - write 0x5 - read 0x5 - match OK
[D] CONFIG_TDP_CONTROL - write 0x0 - read 0x0 - match OK
[D] MSR PACKAGE_POWER_LIMIT - write 0x42816000dd8160 - read 0x42816000dd8160 - match OK
[D] MCHBAR PACKAGE_POWER_LIMIT - write 0x42816000dd8160 - read 0xffffffffffffffff - match ERR

mpawlowski-eyeo, 2022-08-01:
> I created a pull request + a fork where I bypassed the check so that we can run some tests. If you wanna run it, simply compile it using the original manual.

I built your change and my results:

During multi-core heavy load (compiling Chromium) the CPUs spool up to ~2500 MHz and temps go up to 95 deg. C for a few moments, then the CPUs get throttled down to 1000 MHz and temps fall to 60 deg (and stay there).

Feels like I'm reaching some tripping point (sustained watts?). 

Without `throttled`, the temps hover around 80 deg C, performance cores settle at 2200 MHz and efficiency cores at 700 MHz.

i7-1270p is that weird architecture where you have two different kinds of cores - is this handled by `throttled`?

mkogan1, 2022-08-01:
@mpawlowski-eyeo 
check if the tcc offset is increasing (in example below it's 1 deg C but on my laptop jumps under load to 20 deg C or more)
```
sudo turbostat -n 1 -i 1 |& grep offset
cpu0: MSR_IA32_TEMPERATURE_TARGET: 0x01640000 (99 C) (100 default - 1 offset)
```

lakotamm, 2022-08-01:
> I built your change and my results:
>
> During multi-core heavy load (compiling Chromium) the CPUs spool up to ~2500 MHz and temps go up to 95 deg. C for a few moments, then the CPUs get throttled down [...]

Could you run throttled.py with --debug flag and post the output?

I am wondering whether MCHBAR pkg. power limit failing to synchronize with MSR pkg. power limit is one of the culprits.

lakotamm, 2022-08-02:
> 

> `sudo turbostat -n 1 -i 1 |& grep offset`

This command does not return anything on i7-12700H. Turbostat does not report any offset on my i7-12700H.
```
cpu0: MSR_IA32_TEMPERATURE_TARGET: 0x08640000 (100 C)
cpu0: MSR_IA32_PACKAGE_THERM_STATUS: 0x883d0000 (39 C)
cpu0: MSR_IA32_PACKAGE_THERM_INTERRUPT: 0x02000003 (100 C, 100 C)
```

mpawlowski-eyeo, 2022-08-02:
```
$ sudo ./throttled/throttled.py --debug
[sudo] password for mpawlowski: 
[I] Detected CPU architecture: Intel AlderLake
[I] Trying to unlock MSR allow_writes.
[I] Testing if undervolt is supported...
[I] Testing if HWP is supported...
[I] Loading config file.
[D] cpu platform info: maximum non turbo ratio = 25
[D] cpu platform info: maximum efficiency ratio = 4
[D] cpu platform info: minimum operating ratio = 4
[D] cpu platform info: feature ppin cap = 1
[D] cpu platform info: feature programmable turbo ratio = 1
[D] cpu platform info: feature programmable tdp limit = 1
[D] cpu platform info: number of additional tdp profiles = 2
[D] cpu platform info: feature programmable temperature target = 1
[D] cpu platform info: feature low power mode = 1
[D] Undervolt plane CORE - write 0 mV (0x0) - read 0 mV (0x0) - match OK
[D] Undervolt plane GPU - write 0 mV (0x0) - read 0 mV (0x0) - match OK
[D] Undervolt plane CACHE - write 0 mV (0x0) - read 0 mV (0x0) - match OK
[D] Undervolt plane UNCORE - write 0 mV (0x0) - read 0 mV (0x0) - match OK
[D] Undervolt plane ANALOGIO - write 0 mV (0x0) - read 0 mV (0x0) - match OK
[I] Starting main loop.
[D] core 0 thermal status: thermal limit status = 0
[D] core 0 thermal status: thermal limit log = 0
[D] core 0 thermal status: prochot or forcepr status = 0
[D] core 0 thermal status: prochot or forcepr log = 0
[D] core 0 thermal status: crit temp status = 0
[D] core 0 thermal status: crit temp log = 0
[D] core 0 thermal status: thermal threshold1 status = 0
[D] core 0 thermal status: thermal threshold1 log = 0
[D] core 0 thermal status: thermal threshold2 status = 0
[D] core 0 thermal status: thermal threshold2 log = 0
[D] core 0 thermal status: power limit status = 0
[D] core 0 thermal status: power limit log = 1
[D] core 0 thermal status: current limit status = 0
[D] core 0 thermal status: current limit log = 1
[D] core 0 thermal status: cross domain limit status = 0
[D] core 0 thermal status: cross domain limit log = 0
[D] core 0 thermal status: cpu temp = 60
[D] core 0 thermal status: temp resolution = 1
[D] core 0 thermal status: reading valid = 1
[D] core 1 thermal status: thermal limit status = 0
[D] core 1 thermal status: thermal limit log = 0
[D] core 1 thermal status: prochot or forcepr status = 0
[D] core 1 thermal status: prochot or forcepr log = 0
[D] core 1 thermal status: crit temp status = 0
[D] core 1 thermal status: crit temp log = 0
[D] core 1 thermal status: thermal threshold1 status = 0
[D] core 1 thermal status: thermal threshold1 log = 0
[D] core 1 thermal status: thermal threshold2 status = 0
[D] core 1 thermal status: thermal threshold2 log = 0
[D] core 1 thermal status: power limit status = 0
[D] core 1 thermal status: power limit log = 1
[D] core 1 thermal status: current limit status = 0
[D] core 1 thermal status: current limit log = 1
[D] core 1 thermal status: cross domain limit status = 0
[D] core 1 thermal status: cross domain limit log = 0
[D] core 1 thermal status: cpu temp = 60
[D] core 1 thermal status: temp resolution = 1
[D] core 1 thermal status: reading valid = 1
[D] core 2 thermal status: thermal limit status = 0
[D] core 2 thermal status: thermal limit log = 0
[D] core 2 thermal status: prochot or forcepr status = 0
[D] core 2 thermal status: prochot or forcepr log = 0
[D] core 2 thermal status: crit temp status = 0
[D] core 2 thermal status: crit temp log = 0
[D] core 2 thermal status: thermal threshold1 status = 0
[D] core 2 thermal status: thermal threshold1 log = 1
[D] core 2 thermal status: thermal threshold2 status = 0
[... 237 lines cut ...]
[D] core 14 thermal status: reading valid = 1
[D] core 15 thermal status: thermal limit status = 0
[D] core 15 thermal status: thermal limit log = 0
[D] core 15 thermal status: prochot or forcepr status = 0
[D] core 15 thermal status: prochot or forcepr log = 0
[D] core 15 thermal status: crit temp status = 0
[D] core 15 thermal status: crit temp log = 0
[D] core 15 thermal status: thermal threshold1 status = 0
[D] core 15 thermal status: thermal threshold1 log = 0
[D] core 15 thermal status: thermal threshold2 status = 0
[D] core 15 thermal status: thermal threshold2 log = 0
[D] core 15 thermal status: power limit status = 0
[D] core 15 thermal status: power limit log = 1
[D] core 15 thermal status: current limit status = 0
[D] core 15 thermal status: current limit log = 1
[D] core 15 thermal status: cross domain limit status = 0
[D] core 15 thermal status: cross domain limit log = 0
[D] core 15 thermal status: cpu temp = 62
[D] core 15 thermal status: temp resolution = 1
[D] core 15 thermal status: reading valid = 1
[D] TEMPERATURE_TARGET - write 0xf - read 0xf - match OK
[D] CONFIG_TDP_CONTROL - write 0x0 - read 0x0 - match OK
[D] MSR PACKAGE_POWER_LIMIT - write 0x42816000dd80e8 - read 0x42816000dd80e8 - match OK
[D] MCHBAR PACKAGE_POWER_LIMIT - write 0x42816000dd80e8 - read 0xffffffffffffffff - match ERR
```

mpawlowski-eyeo, 2022-08-02:
> sudo turbostat -n 1 -i 1 |& grep offset

This command also doesn't return anything for me, even under load and after the cores have been throttled down

```
cpu0: MSR_IA32_TEMPERATURE_TARGET: 0x0f640000 (100 C)
cpu0: MSR_IA32_PACKAGE_THERM_STATUS: 0x88360800 (46 C)
cpu0: MSR_IA32_PACKAGE_THERM_INTERRUPT: 0x00000003 (100 C, 100 C)
```

lakotamm, 2022-08-02:
Bingo!
I went through the code and I am 95% sure that I found the reason why it does not work.

Both @mpawlowski-eyeo and I got this error:
`[D] MCHBAR PACKAGE_POWER_LIMIT - write 0x42816000dd80e8 - read 0xffffffffffffffff - match ERR`

In the code I found that `MCHBAR PACKAGE_POWER_LIMIT`is being written into address `0xFED159A0`.  This is a combination of 0xFED10000 (MCHBAR address) + 59A0h offset and it is correct for my 8th gen CPU. However on my 12th gen CPU, using command
`sudo setpci -s 0:0.0 48.l` I can see that MCHBAR address is actually 0xFEDC0001.

So the correct address should actually be 0xFEDC59A0. 

On the top of that it could be good to check whether we are writing correct data into the 16th bit of the register, since its function has been changed in 11th and 12th gen compared to 10th gen CPUs
[
10th gen DATASHEET](https://www.intel.com/content/dam/www/public/us/en/documents/datasheets/10th-gen-core-families-datasheet-vol-2-datasheet.pdf)
[11th gen DATASHEET](https://www.intel.com/content/dam/www/public/us/en/documents/datasheets/10th-gen-core-families-datasheet-vol-2-datasheet.pdf)
[12th gen DATASHEET](https://d2pgu9s4sfmw1s.cloudfront.net/UAM/Prod/Done/a062E00001ULmoxQAD/bec9afd0-eb3e-4583-95b0-e0f1c229a57a?Expires=1659480427&Key-Pair-Id=APKAJKRNIMMSNYXST6UA&Signature=p8Oqx5akGrWxLhrgViXEnKIQayggiVzdgbvDSMMEzJeFEMlTPSJ9CAhO8f1tZ66yGUsfmQtUykzXjggsYf-YoYPPlc6wb6mOQ0erV-p-ejjlvjaeBHZJaoVcVnHPxeQxDb4~ie64tr4H08YZRItlOwAEQjuCEZ3GB6aoeb3QwZUgGEQILzk7jVnFOOQhXV1p2vWIvlQ~sbsSKaA~pAUDH04QntoK6Tb2xGjk85NIOUsK9CjEaDNxEsY-AVq4W6Y~anScToqBDDoX8jCfD~fV7xwFyJAvDBSXtFYMwBFioMA9~JHqvHCQ3SfvYtyyTjYyoV2kNAY4Jg80W1qUXRwh9Q__)

lakotamm, 2022-08-02:
And even better news - changing the value actually fixes the error for me!
`
[D] MCHBAR PACKAGE_POWER_LIMIT - write 0x4280f000dd80c8 - read 0x4280f000dd80c8 - match OK`

I created a commit 2629c786a6f4b17a26dfb6d17b8bf3d69e541937 ([browse files link](https://github.com/lakotamm/throttled/tree/2629c786a6f4b17a26dfb6d17b8bf3d69e541937 )) where I temporarily implemented the fix. Since the address is atm just a fixed number, for safety I disabled support for all other than Alder Lake CPUs. @mpawlowski-eyeo could you give it a try?

mpawlowski-eyeo, 2022-08-03:
@lakotamm I can confirm this works for me too, now the cores stay at 95 degrees as long as I want them to. Good find!

lakotamm, 2022-08-03:
I am glad to hear that it is working!
@erpalma would you by any chance have time to implement the fix?

Btw I am wondering whether also the issues related to the 11th gen CPUs are caused by this - #265 since they report the same MCHBAR error.

erpalma, 2022-08-04:
I should be able to implement this fix tomorrow/next week!

lakotamm, 2022-08-04:
> I should be able to implement this fix tomorrow/next week!
> 
> Also, should we call `setpci` instead of just "assuming" the MCHBAR base?

That sounds great!

I would say that it makes sense to do so and it might save us some work of determining which address goes where. But only if it is not a too complicated thing to do. Right now it seems pretty clear that everything up to and incl. 10th gen uses the old MCHBAR address and all 11th and 12th gen cpus use the newer address.

erpalma, 2022-08-05:
Ok, I've pushed these few lines of code. Feedback are welcome ;)

lakotamm, 2022-08-05:
> Ok, I've pushed these few lines of code. Feedback are welcome ;)

I think that the code looks neat! I will test it ASAP.

My only suggestion is that I would by default assign  0xFEDC0001 address also to Tiger Lake CPUs.
```
warning('Trying to guess the MCHBAR address from the CPUID. This MIGHT NOT WORK!')
        if cpuid in ((6, 151, 2), (6, 154, 3)): # I would add also Tiger lake CPUs here
            MCHBAR_BASE = 0xFEDC0001
        else:
            MCHBAR_BASE = 0xFED10001
```

lakotamm, 2022-08-06:
It seems like everything works alright with the latest update.

lakotamm, 2022-08-06:
I added support for Alder Lake -U series and 6 core -S in pull request #310.