erpalma/throttled #283: No longer installs on Debian

alanbur, 2021-12-21:
Debian 11.1, all requirements installed:

`# ./install.sh 
Copying config file
Config file already exists, skipping
Copying systemd service file
Copying core files
Building virtualenv
Requirement already satisfied: wheel in /usr/lib/python3/dist-packages (0.34.2)
Collecting configparser==5.0.2
  Using cached configparser-5.0.2-py3-none-any.whl (19 kB)
Collecting dbus-python==1.2.18
  Using cached dbus-python-1.2.18.tar.gz (578 kB)
Collecting PyGObject==3.42.0
  Using cached PyGObject-3.42.0.tar.gz (716 kB)
  Installing build dependencies ... error
  ERROR: Command errored out with exit status 1:
   command: /usr/bin/python3 /usr/lib/python3/dist-packages/pip install --ignore-installed --no-user --prefix /tmp/pip-build-env-spwywyiv/overlay --no-warn-script-location --no-binary :none: --only-binary :none: -i https://pypi.org/simple -- setuptools wheel pycairo
       cwd: None
  Complete output (107 lines):
  Collecting setuptools
    Using cached setuptools-60.0.2-py3-none-any.whl (952 kB)
  Collecting wheel
    Using cached wheel-0.37.0-py2.py3-none-any.whl (35 kB)
  Collecting pycairo
    Using cached pycairo-1.20.1.tar.gz (344 kB)
    Installing build dependencies: started
    Installing build dependencies: finished with status 'done'
    Getting requirements to build wheel: started
    Getting requirements to build wheel: finished with status 'done'
      Preparing wheel metadata: started
      Preparing wheel metadata: finished with status 'done'
  Building wheels for collected packages: pycairo
    Building wheel for pycairo (PEP 517): started
    Building wheel for pycairo (PEP 517): finished with status 'error'
    ERROR: Command errored out with exit status 1:
     command: /usr/bin/python3 /tmp/tmpzsers5w7_in_process.py build_wheel /tmp/tmp0sxxuvus
         cwd: /tmp/pip-install-5fywwm1l/pycairo_68a59d6f1e9b4b1b978a4c15207ee597
    Complete output (84 lines):
    running bdist_wheel
    running build
    running build_py
    creating build
    creating build/lib.linux-x86_64-3.9
    creating build/lib.linux-x86_64-3.9/cairo
    copying cairo/__init__.py -> build/lib.linux-x86_64-3.9/cairo
    copying cairo/__init__.pyi -> build/lib.linux-x86_64-3.9/cairo
    copying cairo/py.typed -> build/lib.linux-x86_64-3.9/cairo
    running build_ext
    creating build/temp.linux-x86_64-3.9
    creating build/temp.linux-x86_64-3.9/cairo
    x86_64-linux-gnu-gcc -pthread -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -ffile-prefix-map=/build/python3.9-RNBry6/python3.9-3.9.2=. -fstack-protector-strong -Wformat -Werror=format-security -g -fwrapv -O2 -fPIC -DPYCAIRO_VERSION_MAJOR=1 -DPYCAIRO_VERSION_MINOR=20 -DPYCAIRO_VERSION_MICRO=1 -I/usr/include/cairo -I/usr/include/glib-2.0 -I/usr/lib/x86_64-linux-gnu/glib-2.0/include -I/usr/include/pixman-1 -I/usr/include/uuid -I/usr/include/freetype2 -I/usr/include/libpng16 -I/usr/include/python3.9 -c cairo/bufferproxy.c -o build/temp.linux-x86_64-3.9/cairo/bufferproxy.o -Wall -Warray-bounds -Wcast-align -Wconversion -Wextra -Wformat=2 -Wformat-nonliteral -Wformat-security -Wimplicit-function-declaration -Winit-self -Winline -Wmissing-format-attribute -Wmissing-noreturn -Wnested-externs -Wold-style-definition -Wpacked -Wpointer-arith -Wreturn-type -Wshadow -Wsign-compare -Wstrict-aliasing -Wundef -Wunused-but-set-variable -Wswitch-default -Wno-missing-field-initializers -Wno-unused-parameter -Wno-unused-command-line-argument -fno-strict-aliasing -fvisibility=hidden -std=c99
    x86_64-linux-gnu-gcc -pthread -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -ffile-prefix-map=/build/python3.9-RNBry6/python3.9-3.9.2=. -fstack-protector-strong -Wformat -Werror=format-security -g -fwrapv -O2 -fPIC -DPYCAIRO_VERSION_MAJOR=1 -DPYCAIRO_VERSION_MINOR=20 -DPYCAIRO_VERSION_MICRO=1 -I/usr/include/cairo -I/usr/include/glib-2.0 -I/usr/lib/x86_64-linux-gnu/glib-2.0/include -I/usr/include/pixman-1 -I/usr/include/uuid -I/usr/include/freetype2 -I/usr/include/libpng16 -I/usr/include/python3.9 -c cairo/cairomodule.c -o build/temp.linux-x86_64-3.9/cairo/cairomodule.o -Wall -Warray-bounds -Wcast-align -Wconversion -Wextra -Wformat=2 -Wformat-nonliteral -Wformat-security -Wimplicit-function-declaration -Winit-self -Winline -Wmissing-format-attribute -Wmissing-noreturn -Wnested-externs -Wold-style-definition -Wpacked -Wpointer-arith -Wreturn-type -Wshadow -Wsign-compare -Wstrict-aliasing -Wundef -Wunused-but-set-variable -Wswitch-default -Wno-missing-field-initializers -Wno-unused-parameter -Wno-unused-command-line-argument -fno-strict-aliasing -fvisibility=hidden -std=c99
    x86_64-linux-gnu-gcc -pthread -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -ffile-prefix-map=/build/python3.9-RNBry6/python3.9-3.9.2=. -fstack-protector-strong -Wformat -Werror=format-security -g -fwrapv -O2 -fPIC -DPYCAIRO_VERSION_MAJOR=1 -DPYCAIRO_VERSION_MINOR=20 -DPYCAIRO_VERSION_MICRO=1 -I/usr/include/cairo -I/usr/include/glib-2.0 -I/usr/lib/x86_64-linux-gnu/glib-2.0/include -I/usr/include/pixman-1 -I/usr/include/uuid -I/usr/include/freetype2 -I/usr/include/libpng16 -I/usr/include/python3.9 -c cairo/context.c -o build/temp.linux-x86_64-3.9/cairo/context.o -Wall -Warray-bounds -Wcast-align -Wconversion -Wextra -Wformat=2 -Wformat-nonliteral -Wformat-security -Wimplicit-function-declaration -Winit-self -Winline -Wmissing-format-attribute -Wmissing-noreturn -Wnested-externs -Wold-style-definition -Wpacked -Wpointer-arith -Wreturn-type -Wshadow -Wsign-compare -Wstrict-aliasing -Wundef -Wunused-but-set-variable -Wswitch-default -Wno-missing-field-initializers -Wno-unused-parameter -Wno-unused-command-line-argument -fno-strict-aliasing -fvisibility=hidden -std=c99
    x86_64-linux-gnu-gcc -pthread -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -ffile-prefix-map=/build/python3.9-RNBry6/python3.9-3.9.2=. -fstack-protector-strong -Wformat -Werror=format-security -g -fwrapv -O2 -fPIC -DPYCAIRO_VERSION_MAJOR=1 -DPYCAIRO_VERSION_MINOR=20 -DPYCAIRO_VERSION_MICRO=1 -I/usr/include/cairo -I/usr/include/glib-2.0 -I/usr/lib/x86_64-linux-gnu/glib-2.0/include -I/usr/include/pixman-1 -I/usr/include/uuid -I/usr/include/freetype2 -I/usr/include/libpng16 -I/usr/include/python3.9 -c cairo/device.c -o build/temp.linux-x86_64-3.9/cairo/device.o -Wall -Warray-bounds -Wcast-align -Wconversion -Wextra -Wformat=2 -Wformat-nonliteral -Wformat-security -Wimplicit-function-declaration -Winit-self -Winline -Wmissing-format-attribute -Wmissing-noreturn -Wnested-externs -Wold-style-definition -Wpacked -Wpointer-arith -Wreturn-type -Wshadow -Wsign-compare -Wstrict-aliasing -Wundef -Wunused-but-set-variable -Wswitch-default -Wno-missing-field-initializers -Wno-unused-parameter -Wno-unused-command-line-argument -fno-strict-aliasing -fvisibility=hidden -std=c99
    x86_64-linux-gnu-gcc -pthread -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -ffile-prefix-map=/build/python3.9-RNBry6/python3.9-3.9.2=. -fstack-protector-strong -Wformat -Werror=format-security -g -fwrapv -O2 -fPIC -DPYCAIRO_VERSION_MAJOR=1 -DPYCAIRO_VERSION_MINOR=20 -DPYCAIRO_VERSION_MICRO=1 -I/usr/include/cairo -I/usr/include/glib-2.0 -I/usr/lib/x86_64-linux-gnu/glib-2.0/include -I/usr/include/pixman-1 -I/usr/include/uuid -I/usr/include/freetype2 -I/usr/include/libpng16 -I/usr/include/python3.9 -c cairo/enums.c -o build/temp.linux-x86_64-3.9/cairo/enums.o -Wall -Warray-bounds -Wcast-align -Wconversion -Wextra -Wformat=2 -Wformat-nonliteral -Wformat-security -Wimplicit-function-declaration -Winit-self -Winline -Wmissing-format-attribute -Wmissing-noreturn -Wnested-externs -Wold-style-definition -Wpacked -Wpointer-arith -Wreturn-type -Wshadow -Wsign-compare -Wstrict-aliasing -Wundef -Wunused-but-set-variable -Wswitch-default -Wno-missing-field-initializers -Wno-unused-parameter -Wno-unused-command-line-argument -fno-strict-aliasing -fvisibility=hidden -std=c99
    x86_64-linux-gnu-gcc -pthread -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -ffile-prefix-map=/build/python3.9-RNBry6/python3.9-3.9.2=. -fstack-protector-strong -Wformat -Werror=format-security -g -fwrapv -O2 -fPIC -DPYCAIRO_VERSION_MAJOR=1 -DPYCAIRO_VERSION_MINOR=20 -DPYCAIRO_VERSION_MICRO=1 -I/usr/include/cairo -I/usr/include/glib-2.0 -I/usr/lib/x86_64-linux-gnu/glib-2.0/include -I/usr/include/pixman-1 -I/usr/include/uuid -I/usr/include/freetype2 -I/usr/include/libpng16 -I/usr/include/python3.9 -c cairo/error.c -o build/temp.linux-x86_64-3.9/cairo/error [... cut ...] res-python:>=3.6, <4). Command errored out with exit status 1: /usr/bin/python3 /usr/lib/python3/dist-packages/pip install --ignore-installed --no-user --prefix /tmp/pip-build-env-spwywyiv/overlay --no-warn-script-location --no-binary :none: --only-binary :none: -i https://pypi.org/simple -- setuptools wheel pycairo Check the logs for full command output.
ERROR: Could not find a version that satisfies the requirement PyGObject==3.42.0
ERROR: No matching distribution found for PyGObject==3.42.0
`

erpalma, 2021-12-21:
That's quite strange since `https://pypi.org/project/PyGObject/3.42.0/` is available...

Leryan, 2022-01-02:
I just workaround that by creating the virtualenv by hand, as a *non-root user*:

So, after you've ran `sudo ./install.sh` once:

```
python3 -m virtualenv ./venv
. ./venv/bin/activate
pip install -U pip
pip install wheel
pip install -r requirements.txt

sudo rm -rf /opt/throttled/venv
sudo mv ./venv /opt/throttled/

sudo systemctl daemon-reload
sudo systemctl start throttled

# confirm it is running
sudo systemctl status throttled

# enable at boot
sudo systemctl enable throttled

shrx, 2022-01-18:
I solved it this way:

after install.sh fails:

```
sudo pip3 install PyGObject==3.42.0
sudo ./install.sh
``

LinuxOnTheDesktop, 2022-01-30:
@shrx 

Thank you; your method worked for me.

fpeterschmitt, 2022-03-20:
> I solved it this way:
> 
> after install.sh fails:
> 
> ```
> sudo pip3 install PyGObject==3.42.0
> sudo ./install.sh
> ```

this eventually contaminates the libraries in your entire system, i wouldn't do that.

LinuxOnTheDesktop, 2022-03-20:
@fpeterschmitt : how so, please? And, do you have a better workaround? (Still: I myself have stopped using Throttled, because the problems with it became too numerous.)

fpeterschmitt, 2022-03-20:
> how so, please? And, do you have a better workaround?

because it will run as root, and although i haven't verified that it will *actually* write to `/usr/somethingsomething`, i just don't risk it anymore since `pip` handles installs in non-root user's homedir.

For a workaround, it is the message right before shrx's one: https://github.com/erpalma/throttled/issues/283#issuecomment-1003700864