erpalma/throttled #289: Arch Linux installation instructions do not work properly

fsmiamoto, 2022-01-09:
The current README instructions for Installation on Arch Linux using the Community package do not work properly
since the systemd service file is still called `lenovo_fix`.

```sh
$ pacman -Ql throttled | grep service
throttled /usr/lib/systemd/system/lenovo_fix.service
```

I think the proper fix would be to update the package but maybe we can temporarily update the README just to avoid confusion:
```
$ pacman -S throttled
$ sudo systemctl enable --now lenovo_fix.service
```

Let me know what you think and thanks for the awesome work!

iuliandita, 2022-01-11:
Config file, too, is still `/etc/lenovo_fix.conf`.

avalonv, 2022-02-06:
Came here confused about this

yangniao23, 2022-03-07:
You might try installing throttled-git from the AUR.

fsmiamoto, 2022-03-12:
Yup, that would work but my point is about having misleading docs on the README

CSZongzi, 2022-10-14:
This issue should have been fixed after throttled version 0.10.0-1. PR may no longer needed.