erpalma/throttled #291: MCHBAR PL1 is not set correctly by Throttled under Debian Bullseye

fangfufu, 2022-01-16:
Throttled is not changing the MCHBAR PL1 settings. This can be confirmed via sysfs: 
```
root@smithsonian:/home/fangfufu# cat /sys/devices/virtual/powercap/intel-rapl-mmio/intel-rapl-mmio:0/constraint_0_power_limit_uw
20000000
```
I have set it to 60W in ``/etc/throttled.conf``. 

I can increase the limit by running the following in the background: 
```
watch -n 1 'echo 60000000 > /sys/devices/virtual/powercap/intel-rapl-mmio/intel-rapl-mmio:0/constraint_0_power_limit_uw'
```

It would be nice if throttled does this in the background for me. 

I am running Debian Bullseye using the kernel from the backport:
```
fangfufu@smithsonian:~$ uname -a
Linux smithsonian 5.14.0-0.bpo.2-amd64 #1 SMP Debian 5.14.9-2~bpo11+1 (2021-10-10) x86_64 GNU/Linux
```

I can confirm that throttled is working correctly:
```
fangfufu@smithsonian:~$ sudo systemctl status throttled
[sudo] password for fangfufu:
● throttled.service - Stop Intel throttling
     Loaded: loaded (/etc/systemd/system/throttled.service; enabled; vendor preset: enabled)
     Active: active (running) since Sun 2022-01-16 02:10:10 GMT; 33min ago
   Main PID: 115699 (python3)
      Tasks: 2 (limit: 76701)
     Memory: 11.7M
        CPU: 441ms
     CGroup: /system.slice/throttled.service
             └─115699 /opt/throttled/venv/bin/python3 /opt/throttled/throttled.py

Jan 16 02:10:10 smithsonian systemd[1]: Started Stop Intel throttling.
Jan 16 02:10:10 smithsonian python3[115699]: [I] Detected CPU architecture: Intel TigerLake
Jan 16 02:10:10 smithsonian python3[115699]: [I] Trying to unlock MSR allow_writes.
Jan 16 02:10:10 smithsonian python3[115699]: [I] Testing if undervolt is supported...
Jan 16 02:10:10 smithsonian python3[115699]: [I] Testing if HWP is supported...
Jan 16 02:10:10 smithsonian python3[115699]: [I] Loading config file.
Jan 16 02:10:10 smithsonian python3[115699]: [I] Starting main loop.
```

I set the power properly at ``/etc/throttled.conf``
```
[GENERAL]
# Enable or disable the script execution
Enabled: True
# SYSFS path for checking if the system is running on AC power
Sysfs_Power_Path: /sys/class/power_supply/AC*/online
# Auto reload config on changes
Autoreload: True

## Settings to apply while connected to Battery power
[BATTERY]
# Update the registers every this many seconds
Update_Rate_s: 30
# Max package power for time window #1
PL1_Tdp_W: 60
# Time window #1 duration
PL1_Duration_s: 28
# Max package power for time window #2
PL2_Tdp_W: 60
# Time window #2 duration
PL2_Duration_S: 0.002
# Max allowed temperature before throttling
Trip_Temp_C: 85
# Set cTDP to normal=0, down=1 or up=2 (EXPERIMENTAL)
cTDP: 0
# Disable BDPROCHOT (EXPERIMENTAL)
Disable_BDPROCHOT: False

## Settings to apply while connected to AC power
[AC]
# Update the registers every this many seconds
Update_Rate_s: 5
# Max package power for time window #1
PL1_Tdp_W: 60
# Time window #1 duration
PL1_Duration_s: 28
# Max package power for time window #2
PL2_Tdp_W: 60
# Time window #2 duration
PL2_Duration_S: 0.002
# Max allowed temperature before throttling
Trip_Temp_C: 95
# Set HWP energy performance hints to 'performance' on high load (EXPERIMENTAL)
# Uncomment only if you really want to use it
# HWP_Mode: False
# Set cTDP to normal=0, down=1 or up=2 (EXPERIMENTAL)
cTDP: 2
# Disable BDPROCHOT (EXPERIMENTAL)
Disable_BDPROCHOT: False
```

cor3000, 2022-09-29:
Possibly already mentioned earlier somewhere else (maybe it's helpful here in this thread):

My **workaround** is to press the keyboard shortcut FN+h once after each restart.
Even though not (well?) documented for L14 Gen2, this activates the high performance CPU profile and prevents the CPU throttling under load - until the next restart, though.
I've been doing this for the past year as soon as I notice the unwanted CPU throttling kicks under load.
Still the CPU scales down when idle, as expected.

So far I haven't noticed any negative side effects. 

With shortcuts FN+m, FN+l one can switch back to medium or low profile if needed.

fangfufu, 2022-10-02:
@cor3000 , where did you find out about this information?

fangfufu, 2022-10-02:
I am confirming that this is fixed via https://github.com/erpalma/throttled/pull/310

closed 2022-10-02 by fangfufu

cor3000, 2022-10-03:
> @cor3000 , where did you find out about this information? FN+m doesn't seem to do anything for me, but FN+h and FN+l both do.

I don't remember how I eventually put the clues together... however that's what I found in the Lenovo User Manual for L14 Gen2

> **Intelligent cooling**
> The Intelligent Cooling feature helps you adjust fan speed, computer temperature and performance. This feature works in auto mode by default. Press **Fn+T** to switch between manual mode and auto mode.

https://download.lenovo.com/pccbbs/pubs/l14_gen2_l15_gen2/html_en/index.html#t=en%2FIntelligent_cooling_(topic)_T0000783576.html&rhsearch=FN&rhhlterm=FN&rhsyns=%20

However this FN+t shortcut didn't have any (noticeable) effect on my side.

Then I came across the other shortcuts (FN+h/m/l) documented for other Thinkpad Models, and just "gave it a try"

Maybe from here: https://wiki.archlinux.org/title/Lenovo_ThinkPad_X1_Carbon_(Gen_10)#Performance_modes (even though the model doesn't match, the shortcuts seem to work, maybe not the FN+m (as you noticed), which I don't use anyway)

IIRC: I also saw this in one of the Thinkpad user Guides for an older Model, but as of today I can't find it anymore.

If I knew how to trigger these FN shortcuts from command line, that would give me a chance to automate these during startup.
Maybe that could even give a simpler fix for the underlying problem than polling/updating a configuration file in an endless loop.

cor3000, 2022-10-03:
also this gives the same hint: FN+H (something about a "Lap Detection System", amazing what kind of weird features companies have to come up with, due to 'regulation' protecting people from burning themselves, similar to ... "don't put a cat into a microwave")
https://www.reddit.com/r/thinkpad/comments/i4x8i8/tip_for_x1c7_and_perhaps_other_recent_models/