erpalma/throttled #293: install.sh no longer works after  883b20ec79a9908662a5b6c1eeceb4929ef5f9a8

mpawlowski-eyeo, 2022-01-31:
The commit apparently broke the installation script on my machine:
https://github.com/erpalma/throttled/commit/883b20ec79a9908662a5b6c1eeceb4929ef5f9a8

```
$ sudo throttled/install.sh 
throttled/install.sh: 8: [: systemd: unexpected operator
throttled/install.sh: 11: [: systemd: unexpected operator
throttled/install.sh: 14: [: systemd: unexpected operator
Copying config file
Config file already exists, skipping
throttled/install.sh: 37: [: systemd: unexpected operator
throttled/install.sh: 41: [: systemd: unexpected operator
throttled/install.sh: 45: [: systemd: unexpected operator
Copying core files
Building virtualenv
..
```

running ubuntu 20.04

mpawlowski-eyeo, 2022-01-31:
CC @Connor-Frank

Connor-Frank, 2022-01-31:
Apparently the script requires bash. Try running it with bash. If it works, I'll open another pull request to update the hashbang. If it doesn't work, I'll find a way to do string comparisons in /bin/sh. Sorry about that.

Connor-Frank, 2022-01-31:
Update: the problem was `[ "$INIT" == "systemd" ]`. That works in bash, but apparently with systems that have a different symlink to `/bin/sh` the best way is `[ "$INIT" = "systemd"]`. I was unaware of this incompatibility, and I will open a new pull request to correct this.

mpawlowski-eyeo, 2022-02-01:
Thanks for addressing this so quickly!

Connor-Frank, 2022-02-04:
The pull request appears to have been merged. If possible, could you please verify that the script now works on your machine?

mpawlowski-eyeo, 2022-02-04:
Yep, seems to work:
```
Copying config file
Config file already exists, skipping
Copying systemd service file
Copying core files
Building virtualenv
Requirement already satisfied: wheel in ./venv/lib/python3.8/site-packages (0.37.1)
Requirement already satisfied: configparser==5.0.2 in ./venv/lib/python3.8/site-packages (from -r requirements.txt (line 1)) (5.0.2)
Requirement already satisfied: dbus-python==1.2.18 in ./venv/lib/python3.8/site-packages (from -r requirements.txt (line 2)) (1.2.18)
Requirement already satisfied: PyGObject==3.42.0 in ./venv/lib/python3.8/site-packages (from -r requirements.txt (line 3)) (3.42.0)
Requirement already satisfied: pycairo>=1.16.0 in ./venv/lib/python3.8/site-packages (from PyGObject==3.42.0->-r requirements.txt (line 3)) (1.20.1)
Enabling and starting systemd service
All done.
```

closed 2022-02-04 by mpawlowski-eyeo