erpalma/throttled #305: Manjaro Failed to enable unit: Unit file throttled.service does not exist

StevenUS, 2022-06-09:
commands per read me
```
sudo pacman -S throttled
warning: throttled-0.9.2-2 is up to date -- reinstalling
resolving dependencies...
looking for conflicting packages...

Packages (1) throttled-0.9.2-2

Total Installed Size:  0.08 MiB
Net Upgrade Size:      0.00 MiB

:: Proceed with installation? [Y/n] y
(1/1) checking keys in keyring                                                       [#################################################] 100%
(1/1) checking package integrity                                                     [#################################################] 100%
(1/1) loading package files                                                          [#################################################] 100%
(1/1) checking for file conflicts                                                    [#################################################] 100%
(1/1) checking available disk space                                                  [#################################################] 100%
:: Running pre-transaction hooks...
(1/1) Creating Timeshift snapshot before upgrade...
==> skipping timeshift-autosnap due skipRsyncAutosnap in /etc/timeshift-autosnap.conf set to TRUE.
:: Processing package changes...
(1/1) reinstalling throttled                                                         [#################################################] 100%
:: Running post-transaction hooks...
(1/2) Reloading system manager configuration...
(2/2) Arming ConditionNeedsUpdate...
 
sudo systemctl enable --now throttled.service`                                                                                          
Failed to enable unit: Unit file throttled.service does not exist.`
```
System:
Kernel: 5.15.41-1-MANJARO arch: x86_64 bits: 64 compiler: gcc v: 12.1.0
Type: Laptop System: LENOVO product: 20L8S0JY00 v: ThinkPad T480s

CPU:
 Info: model: Intel Core i7-8650U bits: 64 type: MT MCP arch: Coffee Lake

Graphics:
 Device-1: Intel UHD Graphics 620 vendor: Lenovo driver: i915 v: kernel

fraser018, 2022-06-21:
Try the git version package from AUR. Had the same issue but AUR package worked for me

arusahni, 2022-07-12:
The version in the Arch repo pushes a `lenovo_fix.service`.

StevenUS, 2022-07-13:
> Try the git version package from AUR. Had the same issue but AUR package worked for me

seems to work, thanks