erpalma/throttled #306: Support for: Intel(R) Core(TM) i7-1270P

mpawlowski-eyeo, 2022-07-27:
My Lenovo X1 Carbon Gen 10 doesn't seem supported:
```
$ systemctl status throttled
× throttled.service - Stop Intel throttling
     Loaded: loaded (/etc/systemd/system/throttled.service; enabled; vendor preset: enabled)
     Active: failed (Result: exit-code) since Tue 2022-07-26 17:10:20 CEST; 20h ago
    Process: 1278 ExecStart=/opt/throttled/venv/bin/python3 /opt/throttled/throttled.py (code=exited, status=1/FAILURE)
   Main PID: 1278 (code=exited, status=1/FAILURE)
        CPU: 82ms

lip 26 17:10:20 mpawlowski-ThinkPad-X1-Carbon-Gen-10 systemd[1]: Started Stop Intel throttling.
lip 26 17:10:20 mpawlowski-ThinkPad-X1-Carbon-Gen-10 python3[1278]: [E] Your CPU model is not supported.
lip 26 17:10:20 mpawlowski-ThinkPad-X1-Carbon-Gen-10 python3[1278]: Please open a new issue (https://github.com/erpalma/throttled/issues) specifying:
lip 26 17:10:20 mpawlowski-ThinkPad-X1-Carbon-Gen-10 python3[1278]:  - model name
lip 26 17:10:20 mpawlowski-ThinkPad-X1-Carbon-Gen-10 python3[1278]:  - cpu family
lip 26 17:10:20 mpawlowski-ThinkPad-X1-Carbon-Gen-10 python3[1278]:  - model
lip 26 17:10:20 mpawlowski-ThinkPad-X1-Carbon-Gen-10 python3[1278]:  - stepping
lip 26 17:10:20 mpawlowski-ThinkPad-X1-Carbon-Gen-10 python3[1278]: from /proc/cpuinfo.
lip 26 17:10:20 mpawlowski-ThinkPad-X1-Carbon-Gen-10 systemd[1]: throttled.service: Main process exited, code=exited, status=1/FAILURE
lip 26 17:10:20 mpawlowski-ThinkPad-X1-Carbon-Gen-10 systemd[1]: throttled.service: Failed with result 'exit-code'.
```
Can you please add support?
```
$ cat /proc/cpuinfo 
processor       : 0
vendor_id       : GenuineIntel
cpu family      : 6
model           : 154
model name      : 12th Gen Intel(R) Core(TM) i7-1270P
stepping        : 3
microcode       : 0x41e
cpu MHz         : 1146.141
cache size      : 18432 KB
physical id     : 0
siblings        : 16
core id         : 0
cpu cores       : 12
apicid          : 0
initial apicid  : 0
fpu             : yes
fpu_exception   : yes
cpuid level     : 32
wp              : yes
flags           : fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush dts acpi mmx fxsr sse sse2 ss ht tm pbe syscall nx pdpe1gb rdtscp lm constant_tsc art arch_perfmon pebs bts rep_good nopl xtopology nonstop_tsc cpuid aperfmperf tsc_known_freq pni pclmulqdq dtes64 monitor ds_cpl vmx smx est tm2 ssse3 sdbg fma cx16 xtpr pdcm pcid sse4_1 sse4_2 x2apic movbe popcnt tsc_deadline_timer aes xsave avx f16c rdrand lahf_lm abm 3dnowprefetch cpuid_fault epb invpcid_single ssbd ibrs ibpb stibp ibrs_enhanced tpr_shadow vnmi flexpriority ept vpid ept_ad fsgsbase tsc_adjust bmi1 avx2 smep bmi2 erms invpcid rdseed adx smap clflushopt clwb intel_pt sha_ni xsaveopt xsavec xgetbv1 xsaves split_lock_detect avx_vnni dtherm ida arat pln pts hwp hwp_notify hwp_act_window hwp_epp hwp_pkg_req umip pku ospke waitpkg gfni vaes vpclmulqdq tme rdpid movdiri movdir64b fsrm md_clear serialize pconfig arch_lbr flush_l1d arch_capabilities
vmx flags       : vnmi preemption_timer posted_intr invvpid ept_x_only ept_ad ept_1gb flexpriority apicv tsc_offset vtpr mtf vapic ept vpid unrestricted_guest vapic_reg vid ple shadow_vmcs ept_mode_based_exec tsc_scaling usr_wait_pause
bugs            : spectre_v1 spectre_v2 spec_store_bypass swapgs
bogomips        : 4992.00
clflush size    : 64
cache_alignment : 64
address sizes   : 46 bits physical, 48 bits virtual
power management:

...
```

I'm happy to help if something needs to be checked

rakbladsvalsen, 2022-09-20:
@mpawlowski-eyeo Can you confirm if the latest release of throttled fixes the issue for you?

mpawlowski-eyeo, 2022-09-20:
I had to send back the laptop and can't check right now. 

I'm not 100% sure if I kept using the https://github.com/erpalma/throttled/commit/2629c786a6f4b17a26dfb6d17b8bf3d69e541937 proof-of-concept commit or ended up pulling from master after https://github.com/erpalma/throttled/pull/310 got merged. I *think* I pulled from master and it worked.

IMO we can close this ticket and if I find it's not working after I get the laptop again, I'll file a bug.

rakbladsvalsen, 2022-10-27:
> IMO we can close this ticket and if I find it's not working after I get the laptop again, I'll file a bug.

Alright, so I ended up getting a Lenovo laptop and I can confirm throttled works on Arch Linux:

```
[I] Detected CPU architecture: Intel AlderLake-P/H
[I] Trying to unlock MSR allow_writes.
[I] Testing if undervolt is supported...
[I] Testing if HWP is supported...
[I] Loading config file.
[I] Starting main loop.
```

At least on the X1 Yoga (i7-1280P), all cores perform relatively  _stable_  at around ~95 C, consuming ~40 W of power after running `stress` for some time. I was able to see power bursts of up to 66W as well, so I guess throttled is working as expected.