erpalma/throttled #312: asus flow z13 : Really low CPU (12900H) performance in linux

Maziar123, 2022-08-22:
Hi 

I have Problem with ASUS flow z13 CPU performance, same as this seems CPU power limit to **15w** 

[](https://linustechtips.com/topic/1433263-really-low-cpu-12900h-performance-in-linux-manjaro)

lakotamm, 2022-08-22:
Which version of throttled are you using?

Maziar123, 2022-08-23:
use AUR : throttled-git.git

lakotamm, 2022-08-23:
Can you give us the output of --debug?

https://github.com/erpalma/throttled#debug

Maziar123, 2022-08-23:
Seems CPU max frequency liming to 1.2GHZ !

lakotamm, 2022-08-23:
> Can you give us the output of --debug?
> 
> https://github.com/erpalma/throttled#debug

It is very hard from me to help you if you do not provide the outputs of the command.

Maziar123, 2022-08-23:
[trsave1.log](https://github.com/erpalma/throttled/files/9400779/trsave1.log)

kernel param GRUB_CMDLINE_LINUX_DEFAULT="cpuidle.off=1 idle=poll processor.ignore_ppc=1 processor.max_cstate=0 intel_idle.max_cstate=0 msr.allow_writes=on udev.log_priority=3"
[cpupower-freqInfo.txt](https://github.com/erpalma/throttled/files/9400819/cpupower-freqInfo.txt)

[inxi.txt](https://github.com/erpalma/throttled/files/9400817/inxi.txt)

lakotamm, 2022-08-23:
> [trsave1.log](https://github.com/erpalma/throttled/files/9400779/trsave1.log)

According to my understanding from this log it seems like throttled is working fine. It seems like the CPU is being limited by the governor setting.

Do you have  **power-profiles-daemon** installed and running?

`sudo systemctl status power-profiles-daemon`

I do not really know how to controll it in KDE, but in GNOME I have the option to set it to "Power saver" "Balanced" and "Performance" setting. Could you set it to "Balanced" or "Performance"? On "Power saver" it also limits my 12700H to ~1,15Ghz.

I would also suggest that you upgrade to the latest kernel 5.19. 5.15 does not have a good support for Alder Lake.

Maziar123, 2022-08-23:
No, TLP or power-profiles-daemon installed ...

Maziar123, 2022-08-23:
No, TLP or power-profiles-daemon **NOT** installed ...

lakotamm, 2022-08-23:
Ok, can you in that case check that the system check the --monitor output?
https://github.com/erpalma/throttled#monitoring
It should show us reasons for throttling in real time. If everything is on "OK" then the CPU is not throttling and the frequency is limited by the governor.

Maziar123, 2022-08-23:
[I] Detected CPU architecture: Intel AlderLake-P/H
[I] Trying to unlock MSR allow_writes.
[I] Testing if undervolt is supported...
[I] Testing if HWP is supported...
[I] Loading config file.
[I] Starting main loop.
[D] Undervolt offsets: CORE: 0.00 mV | GPU: 0.00 mV | CACHE: 0.00 mV | UNCORE: 0.00 mV | ANALOGIO: 0.00 mV
[D] IccMax: CORE: 160.00 A | GPU: 33.00 A | CACHE: 0.00 A
[D] Realtime monitoring of throttling causes:

[AC] Thermal: OK - Power: OK - **Current: LIM** - Cross-domain (e.g. GPU): OK  ||  VCore: 767 mV - Package: 10.6 W - Graphics: 0.5 W - DRAM: 0.0 W - Total: 11.0[AC] Thermal: OK - Power: OK - Current: LIM - Cross-domain (e.g. GPU): OK  ||  VCore: 780 mV - Package: 8.9 W - Graphics: 0.0 W - DRAM: 0.0 W - Total: 8.9 W[AC] Thermal: OK - Power: OK - Current: LIM - Cross-domain (e.g. GPU): OK  ||  VCore: 779 mV - Package: 9.0 W - Graphics: 0.0 W - DRAM: 0.0 W - Total: 9.0 W[AC] Thermal: OK - Power: OK - Current: LIM - Cross-domain (e.g. GPU): OK  ||  VCore: 779 mV - Package: 9.0 W - Graphics: 0.0 W - DRAM: 0.0 W - Total: 9.1 W[AC] Thermal: OK - Power: OK - Current: LIM - Cross-domain (e.g. GPU): OK  ||  VCore: 780 mV - Package: 9.0 W - Graphics: 0.0 W - DRAM: 0.0 W - Total: 9.0 W[AC] Thermal: OK - Power: OK - Current: LIM - Cross-domain (e.g. GPU): OK  ||  VCore: 766 mV - Package: 8.9 W - Graphics: 0.0 W - DRAM: 0.0 W - Total: 9.0 W[AC] Thermal: OK - Power: OK - Current: LIM - Cross-domain (e.g. GPU): OK  ||  VCore: 766 mV - Package: 9.1 W - Graphics: 0.0 W - DRAM: 0.0 W - Total: 9.2 W[AC] Thermal: OK - Power: OK - Current: LIM - Cross-domain (e.g. GPU): OK  ||  VCore: 767 mV - Package: 9.0 W - Graphics: 0.1 W - DRAM: 0.0 W - Total: 9.1 W[AC] Thermal: OK - Power: OK - Current: LIM - Cross-domain (e.g. GPU): OK  ||  VCore: 767 mV - Package: 8.8 W - Graphics: 0.2 W - DRAM: 0.0 W - Total: 9.0 W

Maziar123, 2022-08-23:
current limit ??

Mnky313, 2023-11-29:
Any update on this?

It's still happening for me on Arch (Kernel 6.6.1).

Same exact behaviour, current limit under any load. stuck at ~10W