erpalma/throttled #316: Systemd service fails to start on Ubuntu 22.04

shrx, 2022-09-26:
Installation process goes fine and does not report any issues:

```
Copying config file
Config file already exists, skipping
Copying systemd service file
Copying core files
Building virtualenv
created virtual environment CPython3.10.6.final.0-64 in 198ms
  creator CPython3Posix(dest=/opt/throttled/venv, clear=False, no_vcs_ignore=False, global=False)
  seeder FromAppData(download=False, pip=bundle, setuptools=bundle, wheel=bundle, via=copy, app_data_dir=/root/.local/share/virtualenv)
    added seed packages: pip==22.2.2, setuptools==65.3.0, wheel==0.37.1
  activators BashActivator,CShellActivator,FishActivator,NushellActivator,PowerShellActivator,PythonActivator
Requirement already satisfied: wheel in /usr/lib/python3/dist-packages (0.34.2)
Requirement already satisfied: configparser==5.2.0 in /usr/local/lib/python3.10/dist-packages (from -r requirements.txt (line 1)) (5.2.0)
Requirement already satisfied: dbus-python==1.2.18 in /usr/local/lib/python3.10/dist-packages (from -r requirements.txt (line 2)) (1.2.18)
Requirement already satisfied: PyGObject==3.42.2 in /usr/local/lib/python3.10/dist-packages (from -r requirements.txt (line 3)) (3.42.2)
Requirement already satisfied: pycairo>=1.16.0 in /usr/lib/python3/dist-packages (from PyGObject==3.42.2->-r requirements.txt (line 3)) (1.20.1)
Enabling and starting systemd service
All done.
```

But systemd reports that service failed to start:

```
throttled.service - Stop Intel throttling
     Loaded: loaded (/etc/systemd/system/throttled.service; enabled; vendor preset: enabled)
     Active: failed (Result: exit-code) since Mon 2022-09-26 13:38:28 CEST; 3s ago
    Process: 860570 ExecStart=/opt/throttled/venv/bin/python3 /opt/throttled/throttled.py (code=exited, statu>
   Main PID: 860570 (code=exited, status=1/FAILURE)
        CPU: 58ms

Sep 26 13:38:28 shrx-T480 systemd[1]: Started Stop Intel throttling.
Sep 26 13:38:28 shrx-T480 python3[860570]: Traceback (most recent call last):
Sep 26 13:38:28 shrx-T480 python3[860570]:   File "/opt/throttled/throttled.py", line 22, in <module>
Sep 26 13:38:28 shrx-T480 python3[860570]:     import dbus
Sep 26 13:38:28 shrx-T480 python3[860570]: ModuleNotFoundError: No module named 'dbus'
Sep 26 13:38:28 shrx-T480 systemd[1]: throttled.service: Main process exited, code=exited, status=1/FAIL>
Sep 26 13:38:28 shrx-T480 systemd[1]: throttled.service: Failed with result 'exit-code'.
```

shrx, 2022-09-26:
Running the script locally works fine:
```
sudo ./throttled.py
[I] Detected CPU architecture: Intel Kabylake
[I] Trying to unlock MSR allow_writes.
[I] Testing if undervolt is supported...
[I] Testing if HWP is supported...
[I] Loading config file.
[I] Starting main loop.
```

maxhoerstr, 2022-10-04:
I get the same error message.

kylebakerio, 2022-11-15:
are you sure you don't have any debug mode stuff enabled? in some modes throttled will produce output that could be interpreted as an error code and could perhaps cause systemd to assume it failed and end it.

the-gadgeteer, 2024-04-25:
I think I figured out this issue. For some reason the requirements (or dbus specifically as indicated by the errors returned at systemctl) aren't being installed in the venv located by default in /opt/throttled/venv. 

To solve this in bash: 
`$ sudo su -` change to root
`# cd /opt/throttled/` move to the default location for throttled
`# source ./venv/bin/activate` activate venv for throttled
`# python3 -m pip install -r requirements.txt` install requirements
`# exit` get out of root, can use sudo for the rest
`$ sudo systemctl stop throttled.service
$ sudo systemctl start throttled.service
$ sudo systemctl status throttled.service` restart service