erpalma/throttled #323: openSuse RPM package repo

xpufx, 2022-12-20:
https://build.opensuse.org/package/show/home:xpufx/throttled

I built this for my own use. It follows the latest git master/main version and automatically packages it if there's a change. Can be installed with zypper or opi. Providing the link in case you'd like to put it in the readme.

hz61p1, 2023-09-23:
@xpufx Thank you for the package.
On TW after installation I had to move `mmio.py` from `/usr/lib/python3.10/site-packages` to `/usr/lib/python3.11/site-packages` and then it worked.

xpufx, 2023-09-24:
@hz61p1 Glad you found the package useful. I will have to look into determining the active python version for the package. I am not very experienced with packaging but I will give it a go. Thanks for letting me know.