erpalma/throttled #327: Undervolt seems not working on i7-12700H at Lenovo P15v G3

xtermi2, 2023-01-18:
It seems undervolting is not working. I configured 
```
[UNDERVOLT.BATTERY]
# CPU core voltage offset (mV)
CORE: -30
# Integrated GPU voltage offset (mV)
GPU: -10
# CPU cache voltage offset (mV)
CACHE: -30
# System Agent voltage offset (mV)
UNCORE: -10
# Analog I/O voltage offset (mV)
ANALOGIO: 0

# All voltage values are expressed in mV and *MUST* be negative (i.e. undervolt)!
[UNDERVOLT.AC]
# CPU core voltage offset (mV)
CORE: -70
# Integrated GPU voltage offset (mV)
GPU: -50
# CPU cache voltage offset (mV)
CACHE: -70
# System Agent voltage offset (mV)
UNCORE: -50
# Analog I/O voltage offset (mV)
ANALOGIO: 0
```
and logs of throttled.service seems OK:
```
● throttled.service - Stop Intel throttling
     Loaded: loaded (/etc/systemd/system/throttled.service; enabled; preset: enabled)
     Active: active (running) since Wed 2023-01-18 23:33:34 CET; 15min ago
   Main PID: 89663 (python3)
      Tasks: 2 (limit: 76716)
     Memory: 11.9M
        CPU: 314ms
     CGroup: /system.slice/throttled.service
             └─89663 /opt/throttled/venv/bin/python3 /opt/throttled/throttled.py

Jan 18 23:33:34 andi-ThinkPad-L490 systemd[1]: Started Stop Intel throttling.
Jan 18 23:33:34 andi-ThinkPad-L490 python3[89663]: [I] Detected CPU architecture: Intel AlderLake-P/H
Jan 18 23:33:34 andi-ThinkPad-L490 python3[89663]: [I] Trying to unlock MSR allow_writes.
Jan 18 23:33:34 andi-ThinkPad-L490 python3[89663]: [I] Testing if undervolt is supported...
Jan 18 23:33:34 andi-ThinkPad-L490 python3[89663]: [I] Testing if HWP is supported...
Jan 18 23:33:34 andi-ThinkPad-L490 python3[89663]: [I] Loading config file.
Jan 18 23:33:34 andi-ThinkPad-L490 python3[89663]: [I] Starting main loop.
Jan 18 23:41:49 andi-ThinkPad-L490 python3[89663]: [I] Reloading changes.
Jan 18 23:43:14 andi-ThinkPad-L490 python3[89663]: [I] Reloading changes.
Jan 18 23:44:29 andi-ThinkPad-L490 python3[89663]: [I] Reloading changes.
```
but `./throttled.py --monitor` shows no undervolt offset
```
[I] Detected CPU architecture: Intel AlderLake-P/H
[I] Trying to unlock MSR allow_writes.
[I] Testing if undervolt is supported...
[I] Testing if HWP is supported...
[I] Loading config file.
[I] Starting main loop.
[D] Undervolt offsets: CORE: 0.00 mV | GPU: 0.00 mV | CACHE: 0.00 mV | UNCORE: 0.00 mV | ANALOGIO: 0.00 mV
[D] IccMax: CORE: 160.00 A | GPU: 55.00 A | CACHE: 0.00 A
[D] Realtime monitoring of throttling causes:

[I] Reloading changes.er: OK - Current: OK - Cross-domain (e.g. GPU): OK  ||  VCore: 957 mV - Package: 56.6 W - Graphics: 0.3 W - DRAM: 0.0 W - Total: 56.8 W
```
any suggestion?

neil1969, 2023-01-22:
Intel 11th and 12th gen processors do not support undervolting at all.

xtermi2, 2023-01-22:
Thanks @neil1969 , but that's a shame.

closed 2023-01-22 by xtermi2