erpalma/throttled #340: i9-13900HX support

adamdicarlo0, 2023-11-16:
```

evertvorster, 2023-11-17:
Yes, please add support for this CPU, it's getting heavily throttled.

adamdicarlo0, 2023-11-17:
From what I'm seeing online, it sure _sounds_ like most PC firmware locks newer CPUs (including this one) from being undervolted. I've tried intel-undervolt and it's unable to undervolt this CPU. I'm on a System76 addw3 (Adder WS) running coreboot, though, so it seems surprising that System76 would disable it. Haven't looked more into it yet.

alexis-belmonte, 2023-11-17:
Can confirm that my CPU won't go past the ~50W TDP barrier, as shown in `turbostat`:

```
# turbostat --quiet --interval 1 --cpu 0-31 --show "PkgWatt","Busy%","Core","CoreTmp"
Core	Busy%	CoreTmp	PkgWatt
-	97.03	64	47.62
0	75.23	63	47.55
0	97.37
4	85.45	61
4	97.66
8	89.96	63
8	95.96
12	94.64	63
12	96.92
16	95.98	64
16	97.12
20	97.21	61
20	96.93
24	97.93	62
24	97.64
28	96.38	61
28	98.34
32	99.70	64
33	99.74	64
34	99.73	64
35	99.67	64
36	99.73	64
37	99.70	64
38	99.66	63
39	99.70	63
40	99.71	64
41	99.49	64
42	99.72	63
43	99.71	63
44	99.06	63
45	99.55	63
46	99.54	63
47	99.62	63
```

I also personally tried to pass ACPI OSI strings, but to no avail -- FYI I've tried `Windows 2022` and `Linux`, both did not do a difference.

[A bug has been opened on the kernel.org mailing list regarding this.](https://bugzilla.kernel.org/show_bug.cgi?id=217976#c28)

evertvorster, 2023-11-18:
The "HX" variant of the i9-13900 CPU is actually physically unlocked, but it is also used in laptops, so there is reason to believe that it should be able to be undervolted. 

As for power consumption, my Sager NP8875 E on powersave mode draws a constant 104W when running s-tui in stress testing mode, and I have seen a 215W draw in poerformance mode yesterday... So 50W is woefully too little power to draw for this CPU.

evertvorster, 2023-11-18:
On aside note, how does one find out what the various powercaps should be for a system and CPU?