erpalma/throttled #346: i7-1355U support

carlosabalde, 2024-01-25:
Is it possible to add support for the following CPU model?

```
model name      : 13th Gen Intel(R) Core(TM) i7-1355U
cpu family      : 6
model           : 186
stepping        : 3
```

carlosabalde, 2024-01-25:
In the meantime, testing this...

```
❯ git diff
diff --git a/throttled.py b/throttled.py
index b9f4c41..0e2b5f7 100755
--- a/throttled.py
+++ b/throttled.py
@@ -170,6 +170,7 @@ supported_cpus = {
     (6, 165, 5): 'Comet Lake-S',
     (6, 166, 0): 'CometLake',
     (6, 167, 1): 'RocketLake',
+    (6, 186, 3): 'RaptorLake-U',
 }
 
 TESTMSR = False
```

motiejus, 2024-03-27:
There are >1 people curious about this specific CPU :) Any update?

motiejus, 2024-03-27:
I did a test myself. With this package the upper frequency went up from 1.8GHz to 2.2GHz. A welcome improvement, but not done.

[image]

Output of `throttled.py --debug`:
[debug.txt](https://github.com/erpalma/throttled/files/14779504/debug.txt)

motiejus, 2024-04-09:
Unfortunately, that didn't work. It went down to 1.7GHz after a couple of days of uptime and stayed at that.

[image]

carlosabalde, 2024-04-09:
I didn't spent much time on this, but for me is working as expected combined with tuning of the platform profile (i.e. `/sys/firmware/acpi/platform_profile`) and the CPU scaling governor (i.e. `/sys/devices/system/cpu/cpu*/cpufreq/scaling_governor`). Sharing just in case it helps.

motiejus, 2024-04-11:
@carlosabalde those didn't help (both were set to "performance").

HOWEVER removing `tlp` fixed the cpufreq issues on my machine! I had tlp enabled for all my laptops since forever for battery management ("stop charging at 80%"), not any more!

ldoktor, 2024-07-22:
Hello folks, I have this CPU and the https://github.com/erpalma/throttled/issues/346#issuecomment-1911006183 made throttled happy to load and work with it. The system seems to perform well.